hneemann/Digital #657: Global generic variables

YulienPohl, 2021-02-20:
It would be helpful to have a global variable that can be initialized (eg. in code block) and used everywhere.

Currently this is only possible in a code block where you set each circuit with addComponent().
It would then be easier to set existing circuit variables.

hneemann, 2021-02-20:
If there were such a global variable, the circuit would depend on the order in which the components are processed. 
This would be a very confusing side effect.

YulienPohl, 2021-02-20:
What if the code block has an internel order.

hneemann, 2021-02-21:
There is, of course, a fixed sequence in which the components are processed, but this is not visible to the user and cannot be influenced. It is even the case that "harmless" operations, such as reordering the inputs or outputs, change this order.
I have to think about it.

hneemann, 2021-02-22:
In the branch globalVars there is a struct `global` which is identical for all components. 
In this struct values can be written in the code components.
Please let me know, if this works for you.

YulienPohl, 2021-02-23:
It works as expected.

However, one could additionally display an information in the code header, which is processed as 1st.
This would solve the problem with the order.

hneemann, 2021-02-23:
I am thinking about allowing only one code component at all, or allowing write access to the global variables only in the first code component. I am not sure yet.

commit c7b91108 (2021-02-23): allows writing to global field only in the first code component; closes #657
closed 2021-02-23 by hneemann