hneemann/Digital #663: Circuit not rendering in full resolution on macOS with Retina screen

nelson2tm, 2021-02-22:
On macOS, when using a retina screen, the app renders in retina resolution except for the circuit part. The circuit renders in a lower resolution than for example the menu on top:
[image]

Selecting everything and moving it around slightly renders the circuit in retina resolution until the mouse button is released:
[image]

Zoomed in:
Normal (circuit looks pixelated):
[image]
Selected and dragging (circuit looks sharp):
[image]

Expected: the circuit should render in high resolution at all times.

macOS version: 10.15.7
Java version: tried on both 1.8.0_232 and 15.0.2

hneemann, 2021-02-22:
This behavior is intentional. If the rendering of the circuit takes too long (>500ms) the render quality is reduced by disabling anti-alias. Actually this should only happen with very large circuits and should prevent the GUI from becoming too sluggish.
Interesting is the question, why on your computer the rendering takes so long, although there are only so few components.

hneemann, 2021-02-22:
Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/).

nelson2tm, 2021-02-22:
It still appears the same as it did before using the pre release (top screenshot in initial post). I tried measuring the render time of the circuit and it seems to be averaging about 10-20 ms ([CircuitComponent.java:888](https://github.com/hneemann/Digital/blob/59a496f36212d6bec50b21c0a7aa6ac0dbd4a1f3/src/main/java/de/neemann/digital/gui/components/CircuitComponent.java#L888)).

As a test I forced antiAlias to be always true in by changing [lines 885-886 in CircuitComponent.java](https://github.com/hneemann/Digital/blob/59a496f36212d6bec50b21c0a7aa6ac0dbd4a1f3/src/main/java/de/neemann/digital/gui/components/CircuitComponent.java#L885-L886). This yields the same pixelated result as in the top screenshot. Forcing it to be false makes it look like the screenshot below, which is different to the initial screenshot. Toggling AntiAlias doesn't seem to influence the resolution at which the circuit renders at so I doubt it is the causing this problem.
[image]

hneemann, 2021-02-22:
That's weird. 
What happens if only a part of the circuit is selected and moved? 
Does the resolution then seem to be different for both parts?

nelson2tm, 2021-02-22:
Only the part that is currently being moved appears in full resolution. [Screen recording](https://user-images.githubusercontent.com/25322615/108773534-c9ec1500-755e-11eb-87eb-84998c45a49b.mp4)

hneemann, 2021-02-23:
In this case, it seems clear what is happening:
The drawing of the circuit is done as follows: All components that do not move are drawn into a bitmap and this is drawn into the window. What moves is then drawn on top of it. This makes the output faster, because only a few components have to be drawn, most of it comes from the bitmap, which is very fast to draw.
To generate this bitmap, the size of the window is requested. 
I guess that the JVM contains some automatism to simplify the handling of the high resolution by downscaling all screen coordinates. When then the window size is queried, a reduced size is returned, the bitmap is thus too small, and the resolution seems to be reduced.
What kind of VM are you using? Is this something from Apple itself.

nelson2tm, 2021-02-23:
I've tried using both AdoptOpenJDK 8 and OpenJDK 15.0.1, neither came pre-installed by Apple.

hneemann, 2021-02-23:
Searching the internet I found out that on macs with retina display even in open jdk there is an automatic scaling by a factor of two.
If you could turn this off, everything would work, because Digital natively supports high resolutions, which is required for the MS Surface machines. The problem arises because the JVM reports wrong/scaled window sizes.

nelson2tm, 2021-03-01:
I tried launching Digital.jar using the `Dsun.java2d.uiScale=1.0` flag. On Windows this properly sets the scaling to 100%. On macOS however it seems to detect scaling is disabled using this flag and tries to scale it up automatically making the contents of the blurry [(screenshot)](https://user-images.githubusercontent.com/25322615/109481290-dd2a4380-7a7c-11eb-832f-47ea1e615466.png). I'm not sure if there is a way to disable this automatic upscaling.

I believe you can simulate how Digital scales on macOS on Windows by starting Digital using  `java -Dsun.java2d.uiScale=2.0 -jar Digital.jar`. Note how the text in the menu bar is sharp while the circuit is pixelated. Just like on macOS, selecting and moving around the circuit makes it sharp.

[image]

hneemann, 2021-03-01:
Thanks for the hint. On my linux machine the `-Dsun.java2d.uiScale=2.0` works too. 
I will see if I can find a workaround for this.

jnoche, 2021-04-14:
The problem here is also happening to me.  I had high resolution when I was using Java 1.8.0, then got lower resolution when I used Java 15.0.1.

(I first mentioned it here: https://github.com/hneemann/Assembler/issues/5#issuecomment-789659363 when I said "Digital does now look a little different (the components now seem to appear in a lower resolution).)

In my Windows 10 system, typing "java -Dsun.java2d.uiScale=2.0 -jar Digital.jar" at the command prompt results in low resolution, and typing "java -Dsun.java2d.uiScale=1.0 -jar Digital.jar" results in high resolution.

Dman1791, 2023-12-12:
This was the only open issue, so I'll throw my hat in despite it being two years old:

I'm having the same issue on Windows 10. I have a 4k screen and 150% UI scaling in Windows. As with nelson2tm, the circuit is only sharp when being moved. This happens even with nothing but an input pin in the circuit.

jnoche's second java command seems to have it render at full resolution, but also makes the UI rather small. For now I'm just running it off of the command line for the better resolution, despite the small UI.