hneemann/Digital #664: mvn install stall on [main] INFO de.neemann.digital.hdl.vhdl2.VHDLCreator - export main

svenn71, 2021-02-25:
When building from source with maven the process comes to a stall sometime after
HDLExample_dig_tb.vhdl has been performed.

This is the last few lines before the stall:
```
[main] INFO de.neemann.digital.hdl.vhdl2.VHDLCreator - export main
-- tested 67 examples
[main] INFO de.neemann.digital.hdl.vhdl2.VHDLCreator - export SignExt.dig
[main] INFO de.neemann.digital.hdl.vhdl2.VHDLCreator - export Control.dig
[main] INFO de.neemann.digital.hdl.vhdl2.VHDLCreator - export ImReg.dig
[main] INFO de.neemann.digital.hdl.vhdl2.VHDLCreator - export WDmuxCtrl.dig
[main] INFO de.neemann.digital.hdl.vhdl2.VHDLCreator - export Branch.dig
[main] INFO de.neemann.digital.hdl.vhdl2.VHDLCreator - export Flags.dig
[main] INFO de.neemann.digital.hdl.vhdl2.VHDLCreator - export PC_HDL.dig
[main] INFO de.neemann.digital.hdl.vhdl2.VHDLCreator - export ASR.dig
[main] INFO de.neemann.digital.hdl.vhdl2.VHDLCreator - export SWAP.dig
[main] INFO de.neemann.digital.hdl.vhdl2.VHDLCreator - export SWAPN.dig
[main] INFO de.neemann.digital.hdl.vhdl2.VHDLCreator - export LSL.dig
[main] INFO de.neemann.digital.hdl.vhdl2.VHDLCreator - export LSR.dig
[main] INFO de.neemann.digital.hdl.vhdl2.VHDLCreator - export ALU.dig
[main] INFO de.neemann.digital.hdl.vhdl2.VHDLCreator - export Register.dig
[main] INFO de.neemann.digital.hdl.vhdl2.VHDLCreator - export ProcessorHDL.dig
[main] INFO de.neemann.digital.hdl.vhdl2.VHDLCreator - export GPI_HDL.dig
[main] INFO de.neemann.digital.hdl.vhdl2.VHDLCreator - export GPO.dig
[main] INFO de.neemann.digital.hdl.vhdl2.VHDLCreator - export main
/tmp/digital_vhdl_21-02-25_11-42_00_2383507704770401424/HDLExample_dig_tb.vhdl
../../src/ieee/v93/numeric_std-body.vhdl:2098:7:@0ms:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
../../src/synopsys/std_logic_arith.vhdl:255:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:238:16:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:275:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:238:16:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:295:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/ieee/v93/numeric_std-body.vhdl:1032:7:@0ms:(assertion warning): NUMERIC_STD.">": metavalue detected, returning FALSE
../../src/ieee/v93/numeric_std-body.vhdl:1184:7:@0ms:(assertion warning): NUMERIC_STD."<": metavalue detected, returning FALSE
../../src/ieee/v93/numeric_std-body.vhdl:2098:7:@0ms:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
../../src/synopsys/std_logic_arith.vhdl:238:16:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:295:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:238:16:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@140ns:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:275:20:@140ns:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:295:20:@140ns:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@140ns:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@140ns:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@540ns:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:275:20:@540ns:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:295:20:@540ns:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@540ns:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@540ns:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/ieee/v93/numeric_std-body.vhdl:1032:7:@540ns:(assertion warning): NUMERIC_STD.">": metavalue detected, returning FALSE
../../src/ieee/v93/numeric_std-body.vhdl:1184:7:@540ns:(assertion warning): NUMERIC_STD."<": metavalue detected, returning FALSE
../../src/synopsys/std_logic_arith.vhdl:255:20:@540ns:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@540ns:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/ieee/v93/numeric_std-body.vhdl:1032:7:@540ns:(assertion warning): NUMERIC_STD.">": metavalue detected, returning FALSE
../../src/ieee/v93/numeric_std-body.vhdl:1184:7:@540ns:(assertion warning): NUMERIC_STD."<": metavalue detected, returning FALSE
../../src/synopsys/std_logic_arith.vhdl:255:20:@540ns:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@540ns:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@630ns:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:275:20:@630ns:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:295:20:@630ns:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@630ns:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@630ns:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).

[main] INFO de.neemann.digital.hdl.vhdl2.VHDLCreator - export main
[main] INFO de.neemann.digital.hdl.vhdl2.VHDLCreator - export main
-- tested 2 examples
[main] INFO de.neemann.digital.hdl.vhdl2.VHDLCreator - export main
```

hneemann, 2021-02-25:
Hmm! I have started these tests many, many times, and never observed anything similar.
But the next test is the test of a VHDL 8 bit multiplier, which takes quite a long time (all the 65536 combinations are tested). 
What about the processor load? Is it possible that the test just runs for a long time?

svenn71, 2021-02-25:
One processor of a 3GHz Xeon has been running 100% for 9 hours now

hneemann, 2021-02-25:
That's too much! On my machine the complete build takes 1.5min. :smile:

Where is the computation time consumed? In the JVM or in GHDL?

To skip the tests:
```
mvn install -DskipTests
```

svenn71, 2021-02-26:
I just add the exit message from the build.
```
[INFO]
[INFO] Results:
[INFO]
[INFO] Tests run: 74, Failures: 0, Errors: 0, Skipped: 0
[INFO]
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  09:49 h
[INFO] Finished at: 2021-02-26T00:34:32+01:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:3.0.0-M1:test (default-test) on project digital: There are test failures.
[ERROR]
[ERROR] Please refer to /home/svennb/work/prg/git/Digital/target/surefire-reports for the individual test results.
[ERROR] Please refer to dump files (if any exist) [date].dump, [date]-jvmRun[N].dump and [date].dumpstream.
[ERROR] The forked VM terminated without properly saying goodbye. VM crash or System.exit called?
[ERROR] Command was /bin/sh -c cd /home/svennb/work/prg/git/Digital && /usr/lib/jvm/java-11-openjdk-amd64/bin/java -javaagent:/home/svennb/.m2/repository/org/jacoco/org.jacoco.agent/0.8
.4/org.jacoco.agent-0.8.4-runtime.jar=destfile=/home/svennb/work/prg/git/Digital/target/jacoco.exec -jar /home/svennb/work/prg/git/Digital/target/surefire/surefirebooter8113070616626852
661.jar /home/svennb/work/prg/git/Digital/target/surefire 2021-02-25T14-45-13_974-jvmRun1 surefire17464117128806066107tmp surefire_017564298917605116461tmp
[ERROR] Error occurred in starting fork, check output in log
[ERROR] Process Exit Code: 1
[ERROR] Crashed tests:
[ERROR] de.neemann.digital.hdl.vhdl2.VHDLSimulatorTest
[ERROR] org.apache.maven.surefire.booter.SurefireBooterForkException: The forked VM terminated without properly saying goodbye. VM crash or System.exit called?
[ERROR] Command was /bin/sh -c cd /home/svennb/work/prg/git/Digital && /usr/lib/jvm/java-11-openjdk-amd64/bin/java -javaagent:/home/svennb/.m2/repository/org/jacoco/org.jacoco.agent/0.8
.4/org.jacoco.agent-0.8.4-runtime.jar=destfile=/home/svennb/work/prg/git/Digital/target/jacoco.exec -jar /home/svennb/work/prg/git/Digital/target/surefire/surefirebooter8113070616626852
661.jar /home/svennb/work/prg/git/Digital/target/surefire 2021-02-25T14-45-13_974-jvmRun1 surefire17464117128806066107tmp surefire_017564298917605116461tmp
[ERROR] Error occurred in starting fork, check output in log
[ERROR] Process Exit Code: 1
[ERROR] Crashed tests:
[ERROR] de.neemann.digital.hdl.vhdl2.VHDLSimulatorTest
[ERROR]         at org.apache.maven.plugin.surefire.booterclient.ForkStarter.fork(ForkStarter.java:670)
[ERROR]         at org.apache.maven.plugin.surefire.booterclient.ForkStarter.run(ForkStarter.java:283)
[ERROR]         at org.apache.maven.plugin.surefire.booterclient.ForkStarter.run(ForkStarter.java:246)
[ERROR]         at org.apache.maven.plugin.surefire.AbstractSurefireMojo.executeProvider(AbstractSurefireMojo.java:1149)
[ERROR]         at org.apache.maven.plugin.surefire.AbstractSurefireMojo.executeAfterPreconditionsChecked(AbstractSurefireMojo.java:991)
[ERROR]         at org.apache.maven.plugin.surefire.AbstractSurefireMojo.execute(AbstractSurefireMojo.java:837)
[ERROR]         at org.apache.maven.plugin.DefaultBuildPluginManager.executeMojo(DefaultBuildPluginManager.java:137)
[ERROR]         at org.apache.maven.lifecycle.internal.MojoExecutor.execute(MojoExecutor.java:210)
[ERROR]         at org.apache.maven.lifecycle.internal.MojoExecutor.execute(MojoExecutor.java:156)
[ERROR]         at org.apache.maven.lifecycle.internal.MojoExecutor.execute(MojoExecutor.java:148)
[ERROR]         at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject(LifecycleModuleBuilder.java:117)
[ERROR]         at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject(LifecycleModuleBuilder.java:81)
[ERROR]         at org.apache.maven.lifecycle.internal.builder.singlethreaded.SingleThreadedBuilder.build(SingleThreadedBuilder.java:56)
[ERROR]         at org.apache.maven.lifecycle.internal.LifecycleStarter.execute(LifecycleStarter.java:128)
[ERROR]         at org.apache.maven.DefaultMaven.doExecute(DefaultMaven.java:305)
[ERROR]         at org.apache.maven.DefaultMaven.doExecute(DefaultMaven.java:192)
[ERROR]         at org.apache.maven.DefaultMaven.execute(DefaultMaven.java:105)
[ERROR]         at org.apache.maven.cli.MavenCli.execute(MavenCli.java:957)
[ERROR]         at org.apache.maven.cli.MavenCli.doMain(MavenCli.java:289)
[ERROR]         at org.apache.maven.cli.MavenCli.main(MavenCli.java:193)
[ERROR]         at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[ERROR]         at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[ERROR]         at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[ERROR]         at java.base/java.lang.reflect.Method.invoke(Method.java:566)
[ERROR]         at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced(Launcher.java:282)
[ERROR]         at org.codehaus.plexus.classworlds.launcher.Launcher.launch(Launcher.java:225)
[ERROR]         at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode(Launcher.java:406)
[ERROR]         at org.codehaus.plexus.classworlds.launcher.Launcher.main(Launcher.java:347)
[ERROR]
[ERROR] -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
```

svenn71, 2021-02-26:
> That's too much! On my machine the complete build takes 1.5min.
I will have to rerun the build with some timing measurement

> Where is the computation time consumed? In the JVM or in GHDL?
I really don't know how to get this numbers from the build

svenn71, 2021-02-26:
Maybe something interesting with my ghdl. Has been running for about an hour now :smile: 
I am following main on ghdl/ghdl on this machine and the last build was yesterday

```
    PID USER      PRI  NI  VIRT   RES△  SHR S CPU% MEM%   TIME+  Command
2823562 svennb     20   0 10872  4308  4024 S  0.0  0.0  0:00.00 ghdl -a --std=02 --ieee=synopsys Multiply8Bit_dig_tb.vhdl
2823564 svennb     20   0  668M  567M 30740 R 98.6  4.7 57:40.39 /usr/local/bin/ghdl1-llvm --std=02 -fsynopsys -P/usr/local/lib/ghdl/ieee/v93/ -P/usr/local/lib/ghdl/ -c -fpic -o Multipl
```
(truncated by screen size)

hneemann, 2021-02-26:
> I really don't know how to get this numbers from the build

Just start the build including the tests, and if the build stalls, see if the java or the GHDL process is consuming the processing power. If a deadlock should be the cause, e.g. no processing power would be consumed at all.

svenn71, 2021-02-26:
> Just start the build including the tests, and if the build stalls, see if the java or the GHDL process is consuming the processing power. If a deadlock should be the cause, e.g. no processing power would be consumed at all.

It is the GHDL process which is consuming all the processing power

hneemann, 2021-02-26:
In this case, the problem should possibly be reported as a bug in the [GHDL project](https://github.com/ghdl/ghdl).

closed 2021-03-02 by hneemann