hneemann/Digital #669: Undefined startup state for NOT-like gates

milandamen, 2021-03-01:
Whenever the circuit simulation is started, there is quite a chance that NOT-like gates will output a 1 before settling to a 0. This can cause flip flops to start with the wrong state, or any logic that depends on a NOT-like gate's output value to activate. The only solution I've found to combat this is to add a Reset gate and an AND gate after every NOT-like gate to combat these signals. The problem with this is that it adds a 1 gate delay for every NOT-like gate throughout the whole circuit, and it also complicates the circuit with these extra gates.

Is there a way to make something like a 0-delay relay, or some other way to circumvent this?

You can reproduce this with any of the following circuits: https://i.imgur.com/Ot3AguK.png

TheWildJarvi, 2021-03-01:
Its good engineering practice to have a reset on your circuit so that when it powers up reset is true for x amount of time before you can start using it.
-
Jarvi

hneemann, 2021-03-02:
This is a fundamental problem and cannot be avoided. The only possibility is the use of a reset circuit.
What I could do would be to implement a reset circuit that invisibly in the background ensures that the circuit starts in a defined state.
However, I decided to provide only a reset input that automatically generates a reset pulse when the circuit is started, so that the user does not always have to press a reset button. The actual reset circuit is left to the user, because I consider this an important aspect of the circuit design.
If e.g. all components with a state can only change it with a clock pulse, this is not a problem, because no clock signal is generated during the startup phase.
But an unclocked RS flip-flop is problematic in this aspect. And the simulator should reflect this.

[image]

milandamen, 2021-03-02:
Thank you two for the advice!

closed 2021-03-02 by milandamen