hneemann/Digital #670: digital.exe does not find adopt open jdk 11

sparry0407, 2021-03-02:
Sorry Helmut, I am sure this has been raised before and you gave me a fixed copy of the exe; but now I cannot find the Issue record or the patched executable.

If you use digital.exe to start digital, it does not correctly recognize some non-Oracle > jdk 8. In particular it does not see Adopt Open JDK 9+.

hneemann, 2021-03-02:
Hmm! I have tested this in a Win 10 machine and it works well in my case.
But before I look at it again: Is this perhaps a duplicate of #373?

sparry0407, 2021-03-02:
No, the error message is "This application requires a Java Runtime Environment 1.8.0" - I have:
openjdk version "11.0.9.1" 2020-11-04
OpenJDK Runtime Environment AdoptOpenJDK (build 11.0.9.1+1)
OpenJDK 64-Bit Server VM AdoptOpenJDK (build 11.0.9.1+1, mixed mode)
This is the only java I have - it is registered on the path and in the registry (options ticked on set up).

hneemann, 2021-03-02:
And you do not launch the EXE file from the cloned distribution folder but actually the exe file that comes with the unpacked distributed ZIP file?

sparry0407, 2021-03-02:
Correct.

hneemann, 2021-03-02:
It is possible to start the simulator with
```
java -jar Digital.jar
```

sparry0407, 2021-03-02:
If I am in the Digital directory, yes.

hneemann, 2021-03-02:
Is the JAVA_HOME environment variable set?
And if not, could you set this environment variable and try again?

sparry0407, 2021-03-02:
It is set:

C:\SgpShared\ownCloud\pyspace repository\CA-GCE-COMPSCI\General\Digital>echo %JAVA_HOME%
C:\Program Files\AdoptOpenJDK\jdk-11.0.9.101-hotspot\

C:\SgpShared\ownCloud\pyspace repository\CA-GCE-COMPSCI\General\Digital>dir "%JAVA_HOME%"
Volume in drive C is Windows
Volume Serial Number is 483A-3B3A

Directory of C:\Program Files\AdoptOpenJDK\jdk-11.0.9.101-hotspot

13/01/2021  15:13    <DIR>          .
13/01/2021  15:13    <DIR>          ..
13/01/2021  15:13    <DIR>          bin
13/01/2021  15:13    <DIR>          conf
13/01/2021  15:13    <DIR>          include
13/01/2021  15:13    <DIR>          jmods
13/01/2021  15:13    <DIR>          legal
13/01/2021  15:13    <DIR>          lib
10/11/2020  09:34             1,467 release
               1 File(s)          1,467 bytes
               8 Dir(s)  320,243,589,120 bytes free

wgaylord, 2021-03-15:
I can confirm this issue also affects OpenJDK 14 on windows as well.

```
openjdk version "14.0.2" 2020-07-14
OpenJDK Runtime Environment AdoptOpenJDK (build 14.0.2+12)
OpenJDK 64-Bit Server VM AdoptOpenJDK (build 14.0.2+12, mixed mode, sharing)
```

hneemann, 2021-03-26:
I have recreated the exe wrappers with the latest launch4j version.
Could someone check if the new wrappers work any better?

[WindowsWrapper.zip](https://github.com/hneemann/Digital/files/6210439/WindowsWrapper.zip)

sparry0407, 2021-03-26:
Initial test worked, with adopt open JDK 11 registered and on the path.

Stephen Parry
Teacher of Computer Science; Data and Remote Learning Manager | Chapeltown Academy
Tel:  0114 245 4803
Email: ***@***.******@***.***>
“Please note my teaching days are Tuesday and Wednesday; My other working days are Monday and Friday but may vary.”
"Due to my own family/work balance, you may get emails from me outside of normal working hours. Please do not feel any pressure to respond outside of your own working pattern."

Chapeltown Academy, Hydra Business Park, Nether Lane, Sheffield, S35 9ZX

[cid:ca09db08-0f00-4cb6-a91f-45f453c0ad86]

[cid:5b683f2b-35b1-48d0-9cb1-0d905667e01d]

hneemann, 2021-03-26:
Does it work to start the simulator by simple clicking on the Digital.jar?

After playing around with the new exe files in a Win10-VM, I tend to remove the exe files completely as they cause more problems than they solve.

sparry0407, 2021-03-26:
In most cases yes: we have the Adopt JDK on the path and in the registry, so double-clicking the jar usually works. If you have downloaded them sometimes you have to unblock the jar before Windows will  load it and sometimes the AV will warn you of “impending doom”. The one thing that the exe gives the students is a consistent icon to look for that appears on the file and the taskbar.

Stephen Parry
Teacher of Computer Science; Data and Remote Learning Manager | Chapeltown Academy
Tel:  0114 245 4803
Email: ***@***.******@***.***>
“Please note my teaching days are Tuesday and Wednesday; My other working days are Monday and Friday but may vary.”
"Due to my own family/work balance, you may get emails from me outside of normal working hours. Please do not feel any pressure to respond outside of your own working pattern."

Chapeltown Academy, Hydra Business Park, Nether Lane, Sheffield, S35 9ZX

[cid:ca09db08-0f00-4cb6-a91f-45f453c0ad86]

[cid:5b683f2b-35b1-48d0-9cb1-0d905667e01d]

brutspark, 2021-03-26:
Hi there, I would like to contribute to Digital and add well known MCU such as ATMega328 support to Digital, I am currently writing the MCU internal registers and gpio.
Please lets have a call I would like to discuss the contribution and also I try to build source code in netbenas 8 but it doesn't build (please help with building)
Thank you, Regards Sergey. PS brutyan17 skype

hneemann, 2021-03-26:
Since you are asking about the build, I assume you want to implement an AVR MCU in java?
I'm afraid that in this case I probably wouldn't merge your code as I think it's too specific. But in this case you can implement the processor as a [plugin](https://github.com/hneemann/digitalCustomComponents).
It would be quite different if you intend to implement an AVR processor in Digital itself as a circuit. In this case I can imagine to include the processor as an example.

Toliak, 2021-04-13:
I had that issue today.

Reinstalling with all features helped me:
[image]
(So, when I rename all another JDK/JRE folders -- it detects AdoptOpenJDK)

However, way to detect JRE is not so clear :)
It does not use `JAVA_HOME` and use **Oracle javapath** instead
[image]

sparry0407, 2021-04-13:
That’s not so much a problem with Digital, as it is with Java on Windows as a whole; Windows has no one easily accessible, consistent and cross-vendor alternates feature such as Linux has. There are no less than four ways to locate the JDK or JRE: system path; JAVA_HOME environment variable; registry entry; symlinks in the C:\ProgramData\Oracle\Java\javapath\ folder.
When we install Adopt we consistently uninstall any other Java runtimes, then install our Adopt stack with all the options ticked. For most programs that works.

[image]<https://user-images.githubusercontent.com/25643405/114606621-27b9f500-9ca4-11eb-9347-4f42a563ad7f.png>

[image]<https://user-images.githubusercontent.com/25643405/114607247-f55cc780-9ca4-11eb-9eb9-c74082740b53.png>

gsmitheidw, 2021-09-13:
I'm having the same issue - I maintain [Digital](https://community.chocolatey.org/packages/digital) for [Chocolatey.org](https://www.chocolatey.org) and I've just pushed update to v0.28  but detecting java is a bit messy.

I've been using the following basic test in powershell which just checks there is at least some java on the system of some sort:

        [bool]$javafound = (Get-Command java -ErrorAction SilentlyContinue).Name -eq 'java.exe'
	        
	        if ($javafound = $false) 
	        {
	        Write-Warning "Note: No Java Java Runtime found, this is required"
	        }

But sadly OpenJDK does not work, the exe is still looking for the Oracle JRE. I see now that  ```java -jar Digital.jar```  would work fine, but it would mean re-writing my code to ditch the the shortcut generating helper from Chocolatey to use some other code.

hneemann, 2021-09-14:
@gsmitheidw 
Please try this new exe wrapper

[Digital.exe.zip](https://github.com/hneemann/Digital/files/7159105/Digital.exe.zip)

and let me know how it works.

gsmitheidw, 2021-09-14:
Hi @hneemann  - unfortunately I can't even extract that, MS Defender is detecting malware in it:

```Trojan:Script/Wacatac.B!ml```

Presumably just a false positive, there's nothing showing up in virustotal for it. But if that's a problem for me, it'll probably be a problem for many others too.

hneemann, 2021-09-14:
@gsmitheidw 
That is surprising! I created the exe file with the go cross compiler on a linux machine without including any external library. I don't think this file contains any malware.
But you are of course right: this error message would appear everywhere.

This seems to be a common problem: [https://github.com/rust-lang/rust/issues/88297](https://github.com/rust-lang/rust/issues/88297) Maybe it will be fixed soon.