hneemann/Digital #675: Set circuit to test row fails when random is used

maltaisn, 2021-03-07:
It looks like test rows get recalculated after being clicked to set the circuit to their values. This usually works, but if the `random` function is used within the test case, it fails. Different values are set each time, never the correct ones.

https://github.com/hneemann/Digital/blob/5584218ffd4103c453d843bd39ea909bc81adff4/src/main/java/de/neemann/digital/gui/components/testing/ValueTableDialog.java#L190-L195

hneemann, 2021-03-07:
Hmm! I'm not sure if it's worth fixing. 
I'm not even sure if it's a problem, because random values are after all random.

closed 2021-03-07 by hneemann

maltaisn, 2021-03-07:
It depends on what you intended the feature to be used for. I use it to set the circuit to values of failed tests to investigate what went wrong in the circuit. If it doesn't set the same values that were used for the test, then I have to set them manually so the feature doesn't serve me much. Thanks for the fix!

commit 08f091d2 (2021-03-07): preserves the test case seed value during a debug session, closes #675