hneemann/Digital #683: Random Value/Short Circuit Anomaly

KA300SE, 2021-03-19:
Hello! I seem to have run into a wall of confusion. Attached is a picture of a circuit I am working on. The red lines on the right side indicate a path backwards from the main bus in which a random value always forms on the bus. I traced the value back until it reaches the bottom multiplexer. The problem is that, when the sim is active, the lower EEPROM module is NOT outputting anything and therefore the value before the multiplexer is in High-Z. I cannot for the life of me figure out where the random value is coming from. I assumed it might be that the multiplexer would give a random value upon initialization so at one point I added a pull-down resistor before the plexer. While this fixed the random value, it did not fix my second issue. The red lines on the left of the picture indicate an input for outputting the counter to the bus. When I enable this line, with nothing else enabled, it presents me with a short circuit error. No other bits are enabled however so I don't understand what is short ciruiting. Perhaps I am missing a rule about components or I am blind to the error. Please review and let me know of your thoughts.

[image]

[DoubleByte.zip](https://github.com/hneemann/Digital/files/6173526/DoubleByte.zip)

KA300SE, 2021-03-19:
I think I realized that the multiplexer must have a buffer before it can tap into the BUS. An oversight on my part...
However, I still don't quite understand the random value problem. Does the multiplexer in fact output a random value in a high-Z state?

hneemann, 2021-03-19:
What do you expect to happen when you leave a real CMOS input unconnected? :smile: 

See also #665 and https://github.com/hneemann/Digital/issues/36#issuecomment-315969098

KA300SE, 2021-03-19:
Indeed, and I appear to have solved the issue by giving it a run-through again. I usually catch these things before resorted to the forum! Oh well, it's a lesson learned haha!

closed 2021-03-19 by KA300SE