hneemann/Digital #684: [Request] Make ability to create test from currect scheme state

Toliak, 2021-03-22:
**Description**

For example I have the scheme with some inputs and outputs enabled.

[image]

I would like to be able to create test case from current scheme state.

Expected created test:

```
A0 A1 A2 A3 B0 B1 B2 B3 S0 S1 S2 S3 F0 F1 F2 F3 AeqB Cn OVR
0 0 0 1 0 0 0 1 1 0 0 0 1 0 0 1 0 0 1
```

**Workaround**

There is a state table, however, it is vertically, so, I cannot just copy-paste it...

*Maybe this feature is already implemented, however, I cannot find it...*

hneemann, 2021-03-22:
It doesn't make as much sense to derive a test case from the circuit under test itself. 
In this case you only fix what the circuit does, but you do not test what the circuit should do.

But it is possible to analyze the circuit (Analysis->Analysis) and then generate test data (File->Export Test Case).
This makes sense e.g. in situations where you want to refactor a circuit and make sure that the current function does not change, regardless of whether it is correct or incorrect.

Toliak, 2021-03-22:
Yes, exactly
I would like to cover my circuit with tests for futher refactoring.

Also I have a problem with *Analysis -> Analysis* when I have a lot of inputs (that expected, because `2^37` is a lot). 
(In my case, I would like to test flags of 16bit operations)

[image]

So, of course I can create a test, however I supposed to paste headers + type values + tests that + fix circuit or fix test data + test again

On the other hand, when I provide `0`,`1` values in simulation, I can view my circuit problems and there will be no dataset failuers... So, this way, I think, is more friendly for circuit test coverage

hneemann, 2021-03-22:
You could embed the circuit under test in a separate test circuit where buses are then generated. With the busses, the tests can then be expressed in a very simple way. And since the test cases would then be much easier to read, they would also be more valuable.

hneemann, 2021-03-24:
closed in aa8c3d8a7dba9827f1a86da2ee36883e22c14c16

closed 2021-03-24 by hneemann