hneemann/Digital #690: Digital Translation, Chinese

hneemann, 2021-03-25:
@itviewer 
in the near future I would like to publish a new release. Therefore the request for updating the translation.
It would be nice if you could complete the translation.

[lang_diff_zh.xml.zip](https://github.com/hneemann/Digital/files/6203852/lang_diff_zh.xml.zip)

itviewer, 2021-03-28:
[lang_diff_zh.zip](https://github.com/hneemann/Digital/files/6217469/lang_diff_zh.zip)

hneemann, 2021-03-28:
@itviewer 
Thank you very much for your effort!

There is a problem with one entry:

en: "Expected {0} but found {1} values in line {2}!"
zh: "行 {2} 期待 {0} 个值"

The {1} parameter is missing, which causes a syntax error.

itviewer, 2021-03-28:
update
[lang_diff_zh.zip](https://github.com/hneemann/Digital/files/6217544/lang_diff_zh.zip)

hneemann, 2021-03-28:
@itviewer 
Thanks again!

commit a2ab7f28 (2021-03-28): updates the Chinese translation, closes #690
closed 2021-03-28 by hneemann