hneemann/Digital #691: Building Digital in Win10

brutspark, 2021-03-26:
Here is the output of Maven in win 10 cmd
C:\Users\bruts\Downloads\Digital-master>mvn verify
[INFO] Scanning for projects...
[INFO]
[INFO] ---------------------< de.neemann.digital:digital >---------------------
[INFO] Building Digital 1.0-SNAPSHOT
[INFO] --------------------------------[ jar ]---------------------------------
[INFO]
[INFO] --- maven-checkstyle-plugin:3.1.0:check (validate) @ digital ---
[INFO] Starting audit...
Audit done.
[INFO]
[INFO] --- git-commit-id-plugin:2.2.1:revision (get-the-git-infos) @ digital ---
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  8.638 s
[INFO] Finished at: 2021-03-26T16:38:25+04:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal pl.project13.maven:git-commit-id-plugin:2.2.1:revision (get-the-git-infos) on project digital: .git directory is not found! Please specify a valid [dotGitDirectory] in your pom.xml -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException

hneemann, 2021-03-26:
During the build process the version number is read from the git repo to be able to display it in the info dialog.
I guess you have downloaded the sources only as a ZIP file?

Please use git clone:

```
git clone https://github.com/hneemann/Digital.git
```

If you don't have git installed yet, you can find it [here](https://git-scm.com/).

brutspark, 2021-03-26:
Dear Helmut I get same error, here is the artifacts.

C:\Users\bruts\Documents\Dev\Digital>mvn verify
[INFO] Scanning for projects...
[INFO]
[INFO] ---------------------< de.neemann.digital:digital >---------------------
[INFO] Building Digital 1.0-SNAPSHOT
[INFO] --------------------------------[ jar ]---------------------------------
[INFO]
[INFO] --- maven-checkstyle-plugin:3.1.0:check (validate) @ digital ---
[INFO] Starting audit...
Audit done.
[INFO]
[INFO] --- git-commit-id-plugin:2.2.1:revision (get-the-git-infos) @ digital ---
[INFO]
[INFO] --- jacoco-maven-plugin:0.8.4:prepare-agent (default-prepare-agent) @ digital ---
[INFO] argLine set to -javaagent:C:\\Users\\bruts\\.m2\\repository\\org\\jacoco\\org.jacoco.agent\\0.8.4\\org.jacoco.agent-0.8.4-runtime.jar=destfile=C:\\Users\\bruts\\Documents\\Dev\\Digital\\target\\jacoco.exec
[INFO]
[INFO] --- maven-resources-plugin:2.6:resources (default-resources) @ digital ---
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] Copying 139 resources
[INFO]
[INFO] --- maven-enforcer-plugin:3.0.0-M2:enforce (enforce-property) @ digital ---
[INFO]
[INFO] --- maven-compiler-plugin:3.1:compile (default-compile) @ digital ---
[INFO] Changes detected - recompiling the module!
[INFO] Compiling 824 source files to C:\Users\bruts\Documents\Dev\Digital\target\classes
[INFO] -------------------------------------------------------------
[ERROR] COMPILATION ERROR :
[INFO] -------------------------------------------------------------
[ERROR] No compiler is provided in this environment. Perhaps you are running on a JRE rather than a JDK?
[INFO] 1 error
[INFO] -------------------------------------------------------------
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  5.065 s
[INFO] Finished at: 2021-03-26T17:00:55+04:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.1:compile (default-compile) on project digital: Compilation failure
[ERROR] No compiler is provided in this environment. Perhaps you are running on a JRE rather than a JDK?
[ERROR]
[ERROR] -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException

C:\Users\bruts\Documents\Dev\Digital>

hneemann, 2021-03-26:
Looks like only a java runtime environment (JRE) is installed. This is only able to run java programs, but not to compile them. Compiling requires a java development kit (JDK).

brutspark, 2021-03-26:
C:\Users\bruts\Documents\Dev\Digital>javac -version
javac 1.8.0_281 

Java JDK I have installed 
C:\Users\bruts\Documents\Dev\Digital>javac -version
javac 1.8.0_281

Toliak, 2021-03-26:
Could you check `C:\Program Files\Java\`? There had to be `jdk1.*` folder...

Also you can check `JAVA_HOME` [environment variable](https://java.com/en/download/help/path.html)

https://stackoverflow.com/questions/19655184/no-compiler-is-provided-in-this-environment-perhaps-you-are-running-on-a-jre-ra

brutspark, 2021-03-26:
OK i solved that issue

closed 2021-03-26 by brutspark