hneemann/Digital #696: VGA component not working

RadiantUwU, 2021-04-07:
i am following this video https://www.youtube.com/watch?v=l7rce6IQDWs
nothing is happening, i even checked if it was getting all signals(it is) and it is not even showing me an error
here's the file 
[worst vga card.dig.txt](https://github.com/hneemann/Digital/files/6274138/worst.vga.card.dig.txt)

hneemann, 2021-04-07:
I don't know whats going on in your circuit, but the Eater VGA card is working fine:

[image]

But it seems to me that it was not taken into account that the VGA clock is four times as high as the circuit clock.

hneemann, 2021-04-14:
Here are some more considerations that may help: A real monitor recognizes the video modes by the polarity of the sync signals and their frequency and pulse durations. A real monitor cannot measure the pixel clock because there is no corresponding clock signal.

Since a frequency measurement is not possible in the simulation, because the circuit does not run in real time, a pixel clock must be provided in the simulation. Frequency and pulse duration of the sync signals are then determined in relation to the pixel clock, and from this the video mode is determined. Therefore, the VGA component must be supplied with a clock four times higher than the clock of the actual circuit to simulate the Ben-Eater VGA card, so that the correct video mode can be derived from the sync signals.

closed 2021-04-14 by hneemann