hneemann/Digital #707: "Too many Inputs used" when Creating circuit for ATF150x CPLD

ProxyPlayerHD, 2021-04-12:
I'm trying to generate a .PLD file for the ATF1502 CPLD using Digital

but for some reason i'm getting the Error mentioned in the title when i try to "create" the circuit in the analysis window.

the CPLD has a total of 32 useable Pins (+4 extra pins if JTAG is disabled), and 4 of those 32 pins are input only, pin 1, 2, 43, and 44.
my circuit is only using a total of 30 IO Pins and i made sure that the input only pins are actually used as inputs.

I saved the .DIG file as a .TXT so i can directly upload it to here, just change the extension back to use it with Digital.
[CPLD Circuit.txt](https://github.com/hneemann/Digital/files/6293790/CPLD.Circuit.txt)

And this is how i actually have the CPLD connected in my KiCad Project. it's a Compact Flash adpater + BIOS extension ROM for old IBM PCs, so i don't have to deal with noisy old harddrives and limit BIOS options.
[image]
the pinout seems correct acording to the [Datasheet](https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-0995-CPLD-ATF1502AS(L)-Datasheet.pdf) but i'll not order the PCBs until i'm certain it's not an issue with the way i hooked up all the inputs/outputs.

TheWildJarvi, 2021-04-12:
When I used the atf1508 chips I would export the design as VHDL, then use prochip designer to fit the pins. Maybe you can try that?

the pinout seems correct acording to the Datasheet but i'll not order the PCBs until i'm certain it's not an issue with the way i hooked up all the inputs/outputs.

hneemann, 2021-04-12:
@ProxyPlayerHD 
Don't order a PCB until you are sure your design fits. If you are using all the pins of the CPLD it might by that you have to swap some pins to make your design fit. The ATF1502 is not "fully connected" which means that swapping pins around may have an affect on fitting. This depends on the complexity of your circuit.

Up to now Digital supports only the IO ports not the Input-Only ones. 
And that's essentially because I only use those in my lab exercises. :smile:

hneemann, 2021-04-12:
@ProxyPlayerHD 
Good new are: Your design fits in ATF1502 PLCC:

[image]

I only had to add pins 1,2 and 44 to the list of allowed pins.

hneemann, 2021-04-12:
@ProxyPlayerHD 
The latest pre [release build](https://infdigital.dhbw-mosbach.de/) supports the pins 1,2,43 and 44.

ProxyPlayerHD, 2021-04-12:
> When I used the atf1508 chips I would export the design as VHDL, then use prochip designer to fit the pins. Maybe you can try that?

I usually did the same except using Verilog and Quartus II to synthesize it for the MAX7000 series CPLDs since they are perfectly pin and binary compatible with the ATF150x CPLDs.

But I wanted to try this feature out for once and didn't want to bother creating a new project for Quartus II.

> @ProxyPlayerHD
> Good new are: Your design fits in ATF1502 PLCC:
> 
> [image]
> 
> I only had to add pins 1,2 and 44 to the list of allowed pins.

I would've been surprised if it didn't fit seeing as I manually wrote almost the same circuit before by hand for WinCUPL. I just don't trust me writing those logic expressions myself without any errors, so being able to use Digital directly is a great help.

Thanks man, I'll try it out later this day.

hneemann, 2021-04-12:
closed in 96b38e45f0f2a6d8d6caa668276dc12723b77379

closed 2021-04-12 by hneemann

ProxyPlayerHD, 2021-04-12:
hmm strange, in the "create" drop-down menu i was able to use the TT2/JEDEC option but when selecting the CUPL one it would still give the "Too many inputs used" error

hneemann, 2021-04-12:
@ProxyPlayerHD 
Please remove the "-" from the signal names.

ProxyPlayerHD, 2021-04-12:
oh i assume CUPL doesn't allow dashes in names then?
hmm i feel like that could use a better worded error message.

also you could add an option to choose between the DIP and PLCC version of the GAL16V8 and GAL22V10? it seems to be hard coded to the PLCC Version and i can't generate a CUPL or TT2 file for it as it's complaing that pin 14 is not an IO Pin, even though it is on the DIP one.

hneemann, 2021-04-12:
see #160

ProxyPlayerHD, 2021-04-13:
awww, well atleast i can copy the Logic Expression and just make the file myself.