hneemann/Digital #709: [UX/Bug] Keyboard layout (language) unexpectedly changes

Toliak, 2021-04-13:
**Description**
I'm frustrated when I open a circuit element properties window, change keyboard layout and, when I close the window, it unexpectedly changes back...

**How to reproduce**
0. Setup multiple language layouts in the OS
1. Create empty circuit, insert any element
2. Open the element properties (RMB)
3. Change keyboard layout
4. Close the properties window
5. Expected: no layout change. Got: layout change

**Screenshots**
Step 1:
[image]

Step 2,3:
[image]

Step 4,5:
[image]

hneemann, 2021-04-13:
I have no idea what is happening, and there is no code in the simulator itself that does this.
I would be surprised if it was even possible in java to change the language of the user interface.
What operating system are you using and does this happen with other Java applications as well?

Toliak, 2021-04-13:
Windows 10 1909
Java version:
```
java version "1.8.0_192"
Java(TM) SE Runtime Environment (build 1.8.0_192-b12)
Java HotSpot(TM) 64-Bit Server VM (build 25.192-b12, mixed mode)
```
I think I don't use other Java apps (except for PyCharm, but it has embedded Java version)

I will check this bug in Manjaro Linux later...

hneemann, 2021-04-13:
Your Java version is quite old.
You should possibly use a more recent version.
[AdoptOpenJDK](https://adoptopenjdk.net/) is a easy to install option.

Toliak, 2021-04-13:
Oracle JDK 15: the same issue
[image]

Manjaro Linux, openjdk 15: no issues (but I use Digital on Windows usually)

AdoptOpenJDK 11: unfortunately, the same issue
[image]

I will try to debug the latest code, maybe will catch something...