hneemann/Digital #710: Cannot find "std" library when checking custom VHDL component

TheWildJarvi, 2021-04-13:
Hi, I am trying to define a custom VHDL component via the external component tool. I am on windows 10 64 bit. I have downloaded GHDL 0.31and installed it. I have GHDL linked in my settings as shown here: 
[image]

I am using this code provided by digikey for a VGA controller: https://www.digikey.com/eewiki/download/attachments/15925278/vga_controller.vhd?version=2&modificationDate=1520459467550&api=v2

After defining the input/Outputs and clicking check, I get the following error: 
[image]

I am not sure what is causing the error. Thanks for your time.
-
Jarvi

YulienPohl, 2021-04-14:
The most likely causes are
1. digital only accepts std_logic and std_logic_vector for 'column' & 'row'
2. digital throws a warning if the label is not the same as the entity name 
3. outdated GHDL version

TheWildJarvi, 2021-04-14:
> 
> 
> The most likely causes are
> 
>     1. digital only accepts std_logic and std_logic_vector for 'column' & 'row'
> 
>     2. digital throws a warning if the label is not the same as the entity name
> 
>     3. outdated GHDL version

Okay 2 and 3 are easy to fix, but in that VHDL code, do you have any advice on how to get rid of the integer use? Can I use a std_logic_vector? Im not sure why digikey would choose to use the integer data type, but I guess that isnt a question for this thread.

hneemann, 2021-04-14:
If you are not particularly interested in this exact code, you can also use the VGA implementation that comes with the processor example.

YulienPohl, 2021-04-14:
> do you have any advice on how to get rid of the integer use?
```vhdl
entity Foo is
    port(x : in STD_LOGIC_VECTOR (31 downto 0));
end Foo;

architecture Bh of Foo is
    signal i : integer;
begin
    s <= to_integer(signed(x));
end Bh;
```

closed 2021-04-28 by TheWildJarvi