hneemann/Digital #711: Paste values into ROMs from external sources

mitolegy, 2021-04-13:
Hi, 
I was just wondering if it is possible to paste data into the ROM or EPROM components using the standard ctrl+c ctrl+v function?
I often use excel to build microcode and being able to directly paste the already compiled microcode directly into Digital would be so much better then hand typing it in. if I remember currently Logisim let you do this but I cant seem to do it in digital.

hneemann, 2021-04-13:
Copy&Paste does not work in this component.
I think the easiest way will be to create a hex file. Such a file is also the easiest to create programmatically. 
Most users will most likely choose this way.

mitolegy, 2021-04-13:
ah you see. that's the issue. I know excel. I'm familiar with it. Its brilliant as you essentially write both code and documentation at the same time in a very human readable format. 

that and I know basically nothing about high level programming. formatting the data in excel into a hex file could maybe be possible. but I think it would require so much formatting it'd just be easier to type it in and deal with the bugs.

would it maybe be possible to add such a function to these components?
the data pasted out of programs like excel already contain a lot of low level formatting in the form of control characters like tab and new line. pasting it out in notepad++ reveals basically the exact same table as it exists in excel. just without the lines between cells. 

food for though I guess. 
this would certainly be one of those quality of life features for us non-programmer plebs who still like circuits.

hneemann, 2021-04-13:
Please try the latest [pre-release](https://infdigital.dhbw-mosbach.de/) and let me know how it works.
Using Libre-Office as a data source works fine, but i have not tested using Excel.

mitolegy, 2021-04-14:
seems to work fine. will stress test it sometime this week when my microcode is done. I'll let you know if I find any issues.
thank you :)

hneemann, 2021-04-19:
closed in c48872d1357796d6de307e58f7349c94e97f57a7

closed 2021-04-19 by hneemann