hneemann/Digital #713: [Enhancement] Ability to provide filepath instead of code in External component

Toliak, 2021-04-18:
**Description**
The present External component UI provides ability to write the code in it.

I would like to suggest the idea: create field for file path and checkbox to select "Use written code / use code from file".

**Advantages**

- User could use an IDE (Visual Studio Code, for example)
- User can open multiple windows with codes
- Code will be stored separately from the main file. That's good for, say, `git` diff view

hneemann, 2021-04-19:
@Toliak 
Please try the branch externalFIle and let me know how it works.
I have added a new component which is based on code stored in a file.

hneemann, 2021-04-23:
@Toliak 
I have added a more radical approach to the problem of locating files during model building and hdl export and removed the ugly `origin` field from the ElementAttributes.
Please try the updated externalFile branch and let me know how it works. 
I'm still thinking about the problem, but so far I haven't come up with a better idea. If I can't think of anything better, I will merge this branch.

Toliak, 2021-04-24:
Awesome. It works as expected

commit f855750e (2021-04-25): updates the release notes, closes #713
closed 2021-04-25 by hneemann