hneemann/Digital #714: [Question/Bug] Export program memory (ROM) to VHDL/Verilog

Toliak, 2021-04-18:
**Description**
I created a simple circuit with program memory (ROM):
[image]

And provided bin file with (not blank) program memory:
[image]

When I would like to export the circuit to VHDL or Verilog, I recieve the blank ROM.
Expected to recieve Verilog with ROM data from the bin file.

Export to Verilog result (no ROM data here :c ):
[image]

hneemann, 2021-04-18:
You are right! The program memory preload is not supported in HDL export.
But loading the data to the ROM directly solves this issue.

Toliak, 2021-04-19:
I there a way to load the data into the ROM (directly) from provided `.bin` file?

hneemann, 2021-04-19:
The ROM data editor has a file menu where you can load data from a file.

Toliak, 2021-04-19:
Oh... I didn't see this menu entrance, thank you!

Toliak, 2021-04-19:
Oops, I've found one mis-enhancement.

In simulation mode (program memory file writes into splitted ROM's):
[image]

Data after imported
[image]

Toliak, 2021-04-20:
So, how can I export to Verilog/VHDL data loaded to all ROMs from a single `bin` file?..

(Without manually spliting binary file into multiple files :) )

hneemann, 2021-04-20:
If you want to load data into different ROM's, you need one file per ROM.

closed 2021-04-20 by Toliak