hneemann/Digital #722: Syntax for tests with multi-bit input and output

tkoenig1, 2021-04-28:
Is there a syntax for creating test cases with multi-bit input and output?

If so, could it be documented?

It not, would it be possible to implement this?

hneemann, 2021-04-28:
There is nothing special about multi-bit inputs or outputs.
The test of your circuit would look like this:

```
X  A Z 
 0 1 3
 1 0 3
 2 0 2
 3 0 2
 4 0 1
 5 0 1
 6 0 1
 7 0 1
 8 0 0
 9 0 0
10 0 0
11 0 0
12 0 0
13 0 0
14 0 0
15 0 0
```

The test case editor has a help button that shows a brief documentation.
In addition, almost all examples contain test cases.

tkoenig1, 2021-04-28:
Ah, thanks. I had not quite understood that it is simply possible to specify an integer instead of individual bits (what I was looking for).

hneemann, 2021-04-28:
If you want to specify individual bits, you can use the binary notation like this: `0b1010`.

closed 2021-05-01 by hneemann