hneemann/Digital #726: 6-bit LUTs

tkoenig1, 2021-04-29:
First, I have just discovered this simulator recently. Thanks for providing such a great tool!

Now a feature request: Currently, LUTs seem to be limited to five bits. A usual size for FPGA LUTs is six bits of data.

Would it be possible to extend the LUTs in Digital to six bit?

hneemann, 2021-04-29:
Simply enter the value 6.
It's just the combo box which offers only the most common values.

closed 2021-05-01 by hneemann

trayres, 2021-12-31:
If I enter 6, the LUT displays as a 2-input LUT; all other values (up to 20, per #726) seem to work. 

Steps to reproduce:
New Diagram
Place LUT
Edit LUT; set any value 2-20 except 6. The correct number of inputs is displayed.

[image]

Edit LUT again: set value to 6. 2 inputs are displayed. 

OS: Windows 10
build git-revision: v0.28

xotmatrix, 2022-01-01:
Reproduced in v0.28. It works as expected in v0.27. I noticed when adding the default LUT in v0.28 the shape has 2 inputs but the dialog says 6.