hneemann/Digital #727: A Java Exception has occured

HAWKbanido, 2021-04-29:
I've had used Digital before and worked normally. Suddenly, I just cannot open it any more. I've tried using de code on to the cmd and this showed:
[image]
I've also tried updating Java.
What can I do?

hneemann, 2021-04-29:
If the Digital.jar file was created correctly, the icons are included. Did you create the jar file yourself and accidentally not include the resources?

Try it from a folder that does not contain special characters or spaces. On my machine it works also in this case, but I want to be sure that it is not because of that.

HAWKbanido, 2021-04-29:
I did not create the jar file myself. I've just extracted the .zip file. Its odd because it was working like a few days ago. Idk what happened or have any idea how to fix it...

hneemann, 2021-04-29:
Have you tried to launch Digital from a folder where the path does not contain special characters or spaces?

HAWKbanido, 2021-04-29:
Yes!

hneemann, 2021-04-29:
Could you please rename the file Digital.jar to Digital.zip and attach the file to this ticket?

HAWKbanido, 2021-04-29:
[Digital.zip](https://github.com/hneemann/Digital/files/6400856/Digital.zip)

hneemann, 2021-04-29:
The file works perfectly. I have no idea what is going on with your machine.

HAWKbanido, 2021-04-29:
Do you think it could be a problem with Java?

hneemann, 2021-04-29:
Could you please try v0.27-34-g5d29331 which is available for download [here](https://infdigital.dhbw-mosbach.de/)?

It does not fix anything but it adds more detail to the exception.

HAWKbanido, 2021-04-29:
[image]

hneemann, 2021-04-30:
It looks like the java process is not allowed to create temporary files because it lacks the rights to do so.
Seems like there is a problem with the permissions.

hneemann, 2021-05-01:
Since this is more of a problem on your machine, and has little to do with the simulator, I am closing this ticket.

closed 2021-05-01 by hneemann