hneemann/Digital #728: Feature request: add results to output for CLI test

phpeterson-usf, 2021-04-29:
I would like to integrate Digital's command-line testing facility with my [test framework](https://github.com/phpeterson-usf/autograder). In order to make that more useful, I would like to print the detailed output in the test results, rather than just ":passed" or ":failed"

Would you consider making a change like printing `TestResult.results` (via `TestResult::getValueTable()`) in `CommandLineTester::execute()`? in [this area](https://github.com/hneemann/Digital/blob/master/src/main/java/de/neemann/digital/cli/CommandLineTester.java#L83) of the code. 

I can submit a PR if needed, but it looks like just adding `out.println(tr.getValueTable());` would do it.

hneemann, 2021-04-30:
This is not that easy, as there are users who depend on the current output. 
But I can add a new option.

hneemann, 2021-04-30:
Does it make any sense to print the table if the test is passed? I would say no.

hneemann, 2021-04-30:
Hmm! But the output created in case of an error using `out.println(tr.getValueTable());` already contains a diff:

```
unnamed: failed (6%)
C Q_3 Q_2 Q_1 Q_0 
0 1 1 1 1 
0 0 0 0 0 
0 0 0 0 1 
0 0 0 1 0 
0 0 0 1 1 
0 0 1 0 0 
0 0 1 0 E: 0 / F: 1  
0 0 1 1 0 
0 0 1 1 1 
0 1 0 0 0 
0 1 0 0 1 
0 1 0 1 0 
0 1 0 1 1 
0 1 1 0 0 
0 1 1 0 1 
0 1 1 1 0 
0 1 1 1 E: 0 / F: 1 

```

phpeterson-usf, 2021-04-30:
Fair enough. The format of the diff was not obvious to me at first glance, but I see it shows the actual value with an E (for error?) in that column, and the expected value with an F (why F?)

phpeterson-usf, 2021-04-30:
Ah ha :-) I can explain that to students if that notation is important. I would somewhat prefer a word but I'm picking nits now. I appreciate your help.

hneemann, 2021-04-30:
> I would somewhat prefer a word

So do I! :smile: 
But it also appears in the GUI in the results table, and there it bothers when the text representation is to long.

commit 9eb1b230 (2021-04-30): adds a verbose mode to the cli test output; closes #728
closed 2021-04-30 by hneemann