hneemann/Digital #733: Wire and Simulation Confusion

Zachdogg1, 2021-05-02:
I have a really quite simple 32 bit ALU I initially created in Logisim, which I then ported over to digital wire for wire. This ALU preforms as intended in logisim, however I cannot even start the simulation in digital as it insists that there is no input on many of my wires. I suspect the problem is that the D flip flops start not holding any value, meant to be fed values by a multiplexer with either the arithmetic values or a preset or reset. For some reason Digital seems to think that there are no inputs on the wires that feed the multiplexer output back to the flip flops. 
Error
[image]
The full Circuit
[image]

The offending area (an error will pop for every wire here
[image]

The (almost) identical circuit simulating mostly as intended in logisim
[image]

hneemann, 2021-05-02:
In contrast to Logisim in Digital a splitter/merger has a direction.
You have to connect at least one red dot (output) to any wire.

It seems to me that the splitter at the bottom has the wrong direction.

Zachdogg1, 2021-05-02:
Changing the rotation of the bottom splitter does not fix the issue, although I suspect that is not actually changing the direction by doing that. There is an output at the very top of the circuit, 32 bit taking in the q from every d flip flop and merging with a splitter

xotmatrix, 2021-05-02:
For the bottom splitter I believe you want:
Input Splitting: 32
Output Splitting: 1*32

Zachdogg1, 2021-05-02:
That fixed it, forgot to check and just blindly copy pasted an earlier 32 bit splitter

closed 2021-05-02 by Zachdogg1