hneemann/Digital #734: [Enhancement] CLI: Test all files in folder

Toliak, 2021-05-02:
**Description**
I would to suggest an idea: to provide an ability to pass directory path into `java -cp Digital.jar CLI test -cirl [forlder_here]`.

**Workaround (but not working)**
The workaround is to write a simple bash script that traverses directory and runs CLI test for every file.
Unfortunately, if there are no tests, CLI test will fail. (*)

I have a question about (*):
Why is lack of tests - a reason to fail? Is that behavior intended?

```
> java -cp Digital.jar CLI test "C:/.../LEDBlock.dig"
no test cases given
Tests have failed.

> echo %errorlevel%
1

```

hneemann, 2021-05-02:
The reason to add tests to a circuit is to make sure that the circuit works correctly.
And one executes the tests to get a proof that the circuit works as required.
If there is no test, it cannot be proven.

Toliak, 2021-05-02:
Ok, I understood. That makes a sense.
Therefore, workaround can be applied for folders traversing

closed 2021-05-06 by hneemann