hneemann/Digital #736: Components close together hard to click on

rj45, 2021-05-05:
Thank you for this amazing program! I have an annoying but not show stopping UX issue:

When I try to right click on components that are close together it almost always pops open the "which one did you want" dialog, even if I zoom in really far. This seems like it's especially troublesome on constants attached to pins close together. In the previous version of Digital this was not nearly as bad and got much worse when I upgraded. I have linux if that matters.

[image]

As you can see in the screenshot, the "Select" dialog is really not helpful. I have no idea which one of the three listed is the one I just right clicked on. And the one I right clicked on is huge, there should be no need to even display this dialog.

hneemann, 2021-05-06:
@rj45 
Thank's for pointing that out!

The latest pre release fixes this issue.

closed 2021-05-06 by hneemann

rj45, 2021-05-06:
Awesome, thank you so much! It works so much better! I also like that it displays the constant's value so I can see which one is which if I do happen to trigger the dialog. Very appreciated!

commit 0b78f80c (2021-05-06): fixes a mouse selection issue; closes #736