hneemann/Digital #742: Allow for datasize and input/output size to differ for RAM

JAicewizard, 2021-05-13:
Currently, it is not possible to (for example) have datasizes of 8 bits, while also doing something like loading the top half of addr1 and the bottom of addr2. (where addr2 = addr1+1).
Maybe the suggested change in the title isnt the best way to achieve this, maybe something like an "sections/data" which indicate how precise you could address the bits within "data bits".

Currently something like this is possible, but you need n RAM modules for n sections/"data size". For storing you would have to  divide the input data among them and vise versa for loading.

My current use-case is wanting to have a data size of 32 bits across the CPU, but I would also like to have the ability to address the bits at 8 bits of precision. (so being able to read/write at offsets of 8 bits instead of 32)

hneemann, 2021-05-13:
There is such a RAM available in Library->RAMs.
There you will find an implementation of a 32-bit RAM that allows byte and 16-bit word access for reading and writing, including non-aligned accesses.

JAicewizard, 2021-05-13:
Seems like the issue was me not being able to see the library. see https://github.com/hneemann/Digital/issues/526#issuecomment-840572976
Under the assumption that there is indeed such a module I will close this

JAicewizard, 2021-05-13:
Going to reopen this, as it is not possible to load program data into this component (which is something I would want).

Whether or not you're going to end up implementing this, I want to thank you for your time and support!

hneemann, 2021-05-13:
You can load your program data to this ram:

1. Open the RAM32bit
2. Uncheck "Modification Locked" in "Edit->Circuit Specific Settings"
3. Open each of the four 8 bit RAMs and check "Advanced->Program Memory"
4. Open your main circuit
5. Check "Edit->Circuit Specific Settings->Advanced->Preload program memory at startup"
6. Select you program in the input field below.

JAicewizard, 2021-05-13:
Ah I figured it out. The hex file needs to have spaces around the bytes for it to work properly with this setup which made it not work (while the single mem module didnt need spaces).

Kind of genius how the hex is distributed among the mem modules, sorry for bothering you with these problems.

hneemann, 2021-05-13:
Never mind! :smile:

closed 2021-05-13 by hneemann