hneemann/Digital #747: Measurement Table F6 shortcut unreliable

xotmatrix, 2021-05-20:
Sometimes pressing F6 doesn't open the Measurement Table. When the Component Tree is visible, sometimes F6 gives focus to the SearchTextField. Some complicated chain of focusing causes the issue to manifest or not.

I'm not a Java programmer but I think I've narrowed down the problem. It appears the JSplitPane object has a couple of default keyboard shortcuts, F6 and F8. From what I've read, I believe it is possible to suppress them.

Main.java, insert at or around line 394
```
split.getActionMap().getParent().remove("toggleFocus");
```
I would test this but setting up JDK has been nightmarish in the past.

hneemann, 2021-05-20:
Thanks for pointing that out!

If you are on Windows AdoptOpenJDK is easy to install.
And if you are not familar with an Java IDE you can install maven.
After that the build is super simple. Just run

```
mvn install
```

in the folder of the `pom.xml` file

xotmatrix, 2021-05-20:
Thanks for the pointers. I'll give it try.

commit 9dc7a704 (2021-05-21): disables F6 in JSplitPane, closes #747
closed 2021-05-21 by hneemann