hneemann/Digital #751: Stuttering when panning

milandamen, 2021-05-24:
When I pan the view (drag while holding right mouse button) the repainting noticeably stutters, and a single CPU core goes to 100% while the pan is ongoing. The effect seems worse when zoomed in than when zoomed all the way out, and is the same regardless whether a simulation is running or not. Disabling grid does not really improve performance much.

I did some quick profiles. Profile of when zoomed in:

[image]

Profile when zoomed out:

[image]

It seems the biggest CPU hog are the SunGraphics2D functions.
Maybe a fix would be to only draw things that are in the viewport?

Circuit used to test: https://darvit.nl/s/32-bit_RISC-V_CPU_milandamen.dig

System specs:
OS: Kubuntu 20.04
CPU: AMD Threadripper 2950x

hneemann, 2021-05-24:
Hmm! On my machine it works perfectly smooth. But there are circuits missing.
Please use the "File->Export->Export to ZIP file" function to create a ZIP file which contains all the required circuits.

milandamen, 2021-05-24:
Here you go: https://darvit.nl/s/32-bit_RISC-V_CPU_milandamen.zip

hneemann, 2021-05-24:
On my machine, panning is still perfectly smooth. Even when I zoom in.
Could it be that you are using software rendering or something similar?

milandamen, 2021-05-24:
That may be possible, how do I check that?

hneemann, 2021-05-31:
This does not seem to be a problem that I can solve.

closed 2021-05-31 by hneemann