hneemann/Digital #752: Verilog inout connections

mixotricha, 2021-05-26:
When bringing in a verilog component I see input and output fields but no bidirectional field for inout wires. I've got a verilog component with a bidirectional bus attached to it. I've also got another verilog component that is the opposite. It comes in with two buses that I would like to connect together in to a single bidirectional bus. Wondering if I have either just not understood or not figured out how to make this work. Thank you for a wonderful program. Digital is awesome :)  

module z80_top_direct_n(
    output wire nM1,
    output wire nMREQ,
    output wire nIORQ,
    output wire nRD,
    output wire nWR,
    output wire nRFSH,
    output wire nHALT,
    output wire nBUSACK,

    input wire nWAIT,
    input wire nINT,
    input wire nNMI,
    input wire nRESET,
    input wire nBUSRQ,

    input wire CLK,
    output wire [15:0] A,
    inout wire [7:0] D
);

hneemann, 2021-05-27:
inout ports are not supported. Neither in Verilog nor in VHDL.
The only exception are inout ports at the top level to control the pins of the FPGA.

mixotricha, 2021-05-27:
Thank you very much for your reply.

I did figure this out and understand more about the Verilog and FPGA as a
result. I never really understood that before.

In my shematic what I did was add a mux on the data lines with an enable
from the appropriate WR pin off the verilog simulation of the processor (
Z80 ).

I did try the bi-directional pin under the Verilog menu but it hurt my
brain!

Thank you once again for this tool.

I tried Logisim on and off over the years. I knew what I wanted to do. But
I could never quite get it to arrive at it.

I was amazed ( and still am amazed ) that the Digital let me pull in the
Verilog like it did. It figured out the names of the pins and such like.

Now I know modelsim or similar would do all this but that is not open
source is it.

Here is what I have built thanks to your great work. Still a work in
progress but how brilliant to have come so far.

[image: board.png]

I would like to contribute to the Digital project and am curious about the
state of the scope.

Does the scope need some working on? I tried the little guy from the menu
and could not quite get him to chooch.

Hooked him up to a clock and got dashed lines rather than square wave I was
expecting?

hneemann, 2021-05-27:
The scope in the IO menu has a very special purpose. It stores values only in case of a trigger edge.

In most cases, you can simply select "Simulation->Show Measurement Graph" in the menu to display the signals.
In Digital you don't have to connect test leads like you do with a real scope. The scope in Digital simply displays every signal that has a name assigned to it.

hneemann, 2021-05-27:
see also [here](https://github.com/hneemann/Digital/issues/184#issuecomment-416790114)

closed 2021-06-04 by hneemann

kopavel, 2023-09-11:
Pull request for solving that https://github.com/hneemann/Digital/pull/1195