hneemann/Digital #756: Can not close output window after stopping simulation

38569459, 2021-05-29:
OS: linux, with i3wm
Digital version: git-revision v0.27

[image]

As you can see in the image above, the simulation stopped but the output screen still there, no button to close or move it.

38569459, 2021-05-29:
I suggest to either add close/move/resize button to output screen or allow output screen to be focusable when clicked.

hneemann, 2021-05-29:
I can't tell why i3wm doesn't show a close button, but all window managers I've tested so far show a close button.

38569459, 2021-05-29:
i3wm normally don't show close button, so either the app must have it or allow the window to be focusable. In this case, the main screen somehow steal focus from the output screen. A quick fix will be add close/move/resize buttons to output screen, since tracking down why the main screen always steal focus is harder, i believe.

hneemann, 2021-05-29:
> A quick fix will be add close/move/resize buttons to output screen, since tracking down why the main screen always steal focus is harder, i believe.

No, it is not. I have implemented it this way.
In the simulator there are some dialogs that do not allow any user interaction. And these can't take the focus either, because that doesn't make sense. That there are window managers out there which in this case don't allow closing the dialog surprised me.

38569459, 2021-05-29:
Then the only option left is to add some buttons to close it. In i3wm, you must focus the window before you can move/resize/close it, it's how i3wm works and there is nothing i can do about it.

hneemann, 2021-05-29:
There is another solution: the dialog should not get the focus automatically, but the user should be able to give the focus to the dialog.
Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/).

38569459, 2021-05-29:
Yes, I'm now able to focus the window and close it. Thank you for quick reply!

closed 2021-05-29 by hneemann