hneemann/Digital #759: Only 64bits allowed in splitter

38569459, 2021-05-31:
I'm building a circus with lots of wires, ex: 16x64bit data bus and tried to use splitter to make the wiring look better, but the splitter only allow 64bits so i can't do that.
[image]

Each of the line above is 64bits data bus.

hneemann, 2021-05-31:
That's right! All components support up to a maximum of 64 bits.

38569459, 2021-05-31:
Can you raise the limit? Since with a very large data bus, 64bits is quite limited.

hneemann, 2021-05-31:
No.
Moreover, it will always be too small with a sufficiently wide data bus. :smile:

38569459, 2021-06-01:
Is there a data bus component in Digital? I want to avoid UI clutter since lots of wires don't look good.

hneemann, 2021-06-02:
There is nothing available that offers more than 64 bits.

closed 2021-06-05 by 38569459