hneemann/Digital #760: Components/custom menu only available after saving new embedded circus

38569459, 2021-05-31:
Currently, the only way for Components/custom to be available is create and save new embedded circus. But that menu should always available since someone can create a new circus in the same folder or copy circus to that folder. A 'create and save new embedded circus' will force Digital to update but it still quite slow compare to having that menu available and click Update.

hneemann, 2021-05-31:
I do not get the point!
The custom menu has a "update" menu entry.

38569459, 2021-05-31:
The point is, the Custom menu only appear/available after saving first embedded circus. But sometime, i copy/create file in that same folder and can't use those circus since the menu is not available.

hneemann, 2021-05-31:
As long as you do not save or load a circuit, there is no file name/folder to search for custom components.
Only when a circuit is saved, there is a folder that can be searched for custom components.

38569459, 2021-05-31:
Well, I created and saved the first circus, the problem is that now i want to use custom circus, which only available through the Custom menu. But the custom menu only available if i click on 'new embedded circus' and save that new circus. But, sometime, user may copy or create circus in that same folder by other mean, i.e through 'new circus' button, not through 'new embedded circus' button or even copy the file from other project. The thing i want is, the Custom menu always available.

hneemann, 2021-05-31:
Please describe in more detail what you did, what happened, and what you expected to happen instead.

38569459, 2021-05-31:
What i did:
- Create new circus A in X folder
- Create another circus B and save it in X folder
- Look for Component/Custom menu

What happened:
- Folder X is created
- Circus A is created and saved
- Circus B is created and saved
- Component/Custom menu is nowhere to be found

What I expected to happen:
- Component/Custom menu is available

hneemann, 2021-05-31:
On my machine it works as expected. 
Can it be that this is the same problem as #550?

38569459, 2021-05-31:
It's working now, so strange, i can't re-product it anymore. Guess i'll close this issue.

closed 2021-05-31 by 38569459