hneemann/Digital #761: Please add the 16x16 icon

yurivict, 2021-05-31:
On FreeBSD the port installs these icons:
```
$ find . -name "icon*"
./work/Digital-0.27/src/main/svg/icon.svg
./work/Digital-0.27/src/main/resources/icons
./work/Digital-0.27/src/main/resources/icons/icon48.ico
./work/Digital-0.27/src/main/resources/icons/icon64.png
./work/Digital-0.27/src/main/resources/icons/icon32.png
./work/Digital-0.27/src/main/resources/icons/icon128.png
./work/Digital-0.27/src/main/resources/icons/icon48.png
```

However, xfce4 doesn't show the icon in its menu because it needs a 16x16 icon.

Reference: https://gitlab.xfce.org/xfce/xfce4-panel/-/issues/497#note_32915

Thanks,
Yuri

hneemann, 2021-05-31:
Make a suggestion for a 16x16 icon.

yurivict, 2021-05-31:
Sorry, I don't have resources.

I am only a port maintainer who noticed that the icon is broken on xfce4.

commit 4cb3cc36 (2021-06-04): adds a smaller icon, closes #761
closed 2021-06-04 by hneemann