hneemann/Digital #772: 74174 -  vhdl file problem

VanderSant, 2021-06-10:
Hi, everyone
I'm using Digital in my Digital systems classes and found a problem in the VHDL 74174 code

- Component:
<p float="left">
  [image]
  [image] 
</p>

- VHDL:
<p float="left">
  [image]
  [image] 
</p>

- How I solve the problem:
put s2 in place of notCL
[image]

After that, start working correctly.

hneemann, 2021-06-10:
I don't know what's going on on your machine. My output is

``` vhdl
architecture Behavioral of main is
  signal s0: std_logic_vector(5 downto 0);
  signal s1: std_logic;
  signal s2: std_logic_vector(5 downto 0);
begin
  s1 <= NOT notCL;
  s0(0) <= D1;
  s0(1) <= D2;
  s0(2) <= D3;
  s0(3) <= D4;
  s0(4) <= D5;
  s0(5) <= D6;
  gate0: entity work.DIG_D_FF_AS_BUS
    generic map (
      Bits => 6)
    port map (
      Set => '0',
      D => s0,
      C => CLK,
      Clr => s1,
      Q => s2);
  Q1 <= s2(0);
  Q2 <= s2(1);
  Q3 <= s2(2);
  Q4 <= s2(3);
  Q5 <= s2(4);
  Q6 <= s2(5);
end Behavioral;
```

which seems correct. Could you please zip and attach the file which causes the problem?

hneemann, 2021-06-11:
@VanderSant 
There must be some unusual side effect that is causing this problem. Without the circuit I can not find the cause.
The easiest way to create the ZIP file is to use "File->Export->Export to Zip".

VanderSant, 2021-06-12:
sorry for the delay, I check here, and for some reason I was using version 0.24, in the newest version the problem is solved o/

hneemann, 2021-06-12:
see #592

closed 2021-06-12 by hneemann