hneemann/Digital #774: Text Search & Replace?

peterjung1, 2021-06-14:
Would it be possible to add a text Search and Replace function for this like Input and Output names and tunnel net names?

This would be a real time saver for building circuits with a lot of similar names

hneemann, 2021-06-14:
For tunnels, there is already a function to rename all that have the same name.
And for inputs and outputs, you can create an input and then copy it. If the name of the copied input or output ends with a number, this number will be incremented when pasting.

peterjung1, 2021-06-14:
Thank you so much for the fast response – much appreciated!

 

I am using this programme to build a logic simulator so that I can check the functions of ISA/SAMA Logic Diagrams.  I was using Logism before and just started using Digital a few weeks ago – it is a really nice upgrade on Logisim and works very well for what I need – thank you for your work on this.

 

One other quick question – I have a very large sub-circuit I am having trouble with – when I embed it I get the error:

 

Error creating the circuit

Pin System Fault in component (path name.dlg)exists twice

 

Is this telling me I have duplicate Inputs and/or outputs?

 

If so is there a quick way to identify such duplicates in the sub-circuit?

 

Thanks

 

Peter

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

hneemann, 2021-06-14:
> One other quick question – I have a very large sub-circuit I am having trouble with – when I embed it I get the error:
>
> Error creating the circuit
> Pin System Fault in component (path name.dlg)exists [...]
I do not quite understand the question. The name _System Fault_ has been given more than once.
Or is the problem to locate the pin in the circuit? Then the circuit must be really huge. :smile:

peterjung1, 2021-06-14:
Thanks again… I just worked it out and now I feel real dumb!

 

I was reading the error as “Pin System Faut” – as in there is a fault with the pin system…..

 

I just realised it is telling me Pin “System Fault” – as in System Fault is the name of the pin… doh!

 

Thanks again… everyday is a school day! 😊

 

https://www.avast.com/antivirus

commit 96e0c860 (2021-06-14): improves an error message; closes #774
closed 2021-06-14 by hneemann