hneemann/Digital #784: Suggestion of adding a module

RadiantUwU, 2021-06-20:
Could you please add a module that can check a 1 bit's wire if its pulled high, pulled low, highz, high, or low?

hneemann, 2021-06-20:
What would this component look like if it were built with real components on a breadboard?

RadiantUwU, 2021-06-20:
Sorry that i didnt read it, i was watching youtube and working on a 65c02 compatible CPU
I do not quite understand the question, in real life, maybe probing the voltage?
I dont really know, maybe you referred to the question "how you want it to look?" or sum like that thats what i thought anyway
[image]
something like this would still be good
pin 0 is marked, if its 0 or 1, if its in high z its 0, if pulled down its 0, or pulled up its a 1
pin 1 is if its in high z
pin 2 is if its pulled down/pulled up
This will quite help me and other people into doing stuff like fuses

RadiantUwU, 2021-06-20:
You could also use that arrow from "pin control" for the design
Hope this helps!

hneemann, 2021-06-20:
> in real life, maybe probing the voltage?

Measuring the voltage will not be enough, because then how can you distinguish high and pulled-high or low and pulled-low?

RadiantUwU, 2021-06-20:
i mean- idk
but i thought of that cool idea of checking if its pulled up or down cause i kinda need that for an embedded circuit

hneemann, 2021-06-21:
But if you need a component that can distinguish between high and pulled-high or low and pulled-low, the question is how you would build this in reality.
And if you then realize that this is a rather complex thing that requires a current measurement and multiple window detector circuits with difficult to determine window limits, it might be better to think about an alternative to this component.

closed 2021-06-26 by hneemann