hneemann/Digital #785: Yosys compatible block ram modules

rj45, 2021-06-26:
Currently there doesn't appear to be any way to create certain memories in Digital that yosys properly infers as block RAM for ICE40 devices. This is likely a limitation of yosys, but I have no idea when they will fix it since I believe they need to do a pretty big memory inference overhaul in order to tackle this. So that could be years in the future. In the meantime it would be absolutely awesome if Digital had some options for block RAMs that do properly infer.

The Block RAM component should properly infer, but there's no way to make it into a ROM, and there's no way to get separated read and write ports. And as far as I have been able to figure out from testing, this is the only memory module that does properly infer to block RAM.

Anyway, the way I currently have to work around this problem is by creating a top level circuit that looks like this:

[image]

I put all the Digital memories in this module with a register after them so they behave like block RAM. I then export the inner module to verilog and I have to manually write a verilog version of the circuit above with the verilog for memories that yosys does properly infer.

There's two issues with doing this: my verilog is rather poor so I spend a lot of time debugging it, and if the memories are used in a deeply nested module, I have to run the address/data lines through the whole hierarchy up to the top adding a lot of extra noise to the design.

I believe the problem is that the output is registered on ICE40 devices. So it only outputs its contents after the clock goes high. Here's an example of a ROM that properly infers to an initialized block RAM:

```verilog
module ice40_rom (
    input             clk_i,
    input      [8:0] addr_i,
    output reg [7:0] data_o
);

reg [7:0] mem [0:511];

initial $readmemh("./romdata.hex",mem);

always @(posedge clk_i) begin
    data_o <= mem[addr_i];
end

endmodule
```

This is the module in the ice40 docs:

[image]

And here's an example of how that could be translated into verilog (I haven't tested this, though):

```verilog
module ice40_dual_access_bram (
    input wire wr_clk,
    input wire wr_en,
    input wire rd_clk,
    input wire rd_en,
    input wire [8:0] rd_addr,
    input wire [8:0] wr_addr,
    input wire [7:0] data_in,
    output reg [7:0] data_out
);
   reg [7:0] memory [0:511];
   integer i;

   initial begin
      for(i = 0; i <= 511; i=i+1) begin
         memory[i] = 8'b0;
      end
   end

   always @(posedge wr_clk) begin
      if (wr_en) begin
         memory[wr_addr] <= data_in;
      end
   end

   always @(posedge rd_clk) begin
      if (rd_en) begin
        data_out <= memory[rd_addr];
      end
   end
endmodule
```

I would love if there was a digital component that produced verilog similar to the above, especially if the initial state of the ram could come from a file similar to the ROM component. But it would be fine if there was two separate components, one for ROM with a registered output, and one for RAM with a registered output and separate read/write ports.

Even better would be for there to be a configuration parameter on the existing RAM/ROM devices to make their output registered.

rj45, 2021-06-27:
I did a tiny bit more digging and it seems like the pattern matching in yosys is extremely specific.

This code doesn't work at all in the generated ROM:

```verilog
  always @ (*) begin
        if (~sel)
            D = 16'hz;
        else if (A > 8'h76)
            D = 16'h0;
        else
            D = my_rom[A];
    end
```

But if I change it to this:

```verilog
  always @ (*) begin
            D = my_rom[A];
  end
```

Then it almost works, but the register that comes after it cannot have an initial state, since then yosys won't merge it with the memory. So I have to remove this from the flip flop code:

```verilog
    initial begin
        state = Default;
    end
```

I _think_ all flip flops in ice40 devices initialize to zero, and all that code is doing is setting the initial state to zero. So it's a bit baffling why it would fail to merge in that case. But I have worked out that the rules are thus:

1. The register / flip flop after the memory device must not have any initial state set, and cannot have any async reset/set circuitry.
2. The memory device cannot have any if clauses in the always block, except to provide a clock enable feature where if that is false it does nothing.
3. The trick of checking if the address is above a certain value and returning zero if that's the case does not seem to synthesize properly. It would be better to generate a for loop in the initial block to fill the rest of memory with zeros and make sure the memory itself is sized as a power of two. This can be worked around by just ensuring the hex file is padded with zeros, but it's a gotcha.

Anyway @hneemann let me know if you want me to try to submit a PR. You'd probably be much more efficient than I would, but maybe if I submit a PR it could get the ball rolling a bit and you could fix it up for me. If I do work on a PR, I would need to know which direction you would want me to go for the solution. Thanks again for such a wonderful program!

hneemann, 2021-06-28:
It is possible to use the External component to define components with verilog. However, the installation of iverilog is required to simulate the circuit.

Another possibility would be to allow the user to customize the tempates used for HDL code generation. In this way, adaptations to special hardware could be made. I will think about this.

rj45, 2021-07-04:
For the external component feature, I am not sure how fast that would run. Say I am building a circuit with 20 or so independent block RAMs, then that would fork 20 processes during simulation. I feel like that's an inferior solution to what I am already doing where I route the memory data/address lines up to the top level and have separate implementations. But I suppose I could put all 20 block RAMs into a single module, and then I could put that module as deep as I want in the hierarchy. Hmm... I will have to think about that.

With the verilog template customization: I could write a pre-processor that just search/replaces the template with one that works with yosys. So I am not sure how useful that would be.

My primary issue, though, is that the block RAM in the FPGA I am using is synchronous, so in order to simulate that properly in Digital I have to put a register after it. That would be very tricky to remove by search/replace. I would need to probably parse the verilog properly to do that. I can't just construct a fancy regex to do it.

And through testing, I have determined that yosys fails to infer the memory as block RAM because it can't combine the register after the memory with the memory itself. It seems like the register has to be in the same `always` block as the memory read/write for some reason.

This is why I suggested the option to make many/all of the memories synchronous with a checkbox in the advanced settings. I guess that would add a clock input to them so maybe that would be complicated. But then any memory could be made synchronous. I think I could work around the other issues with a fancy search/replace script.

hneemann, 2021-07-20:
closed in favor of #796

closed 2021-07-20 by hneemann