hneemann/Digital #788: File as clock / signal input

YMasquel, 2021-07-09:
version: v0.27
Hello, as enhancement,
should it be possible to use a file for input signals input with : 
 -  timestamp (or delay from origin) ;
 -  level (high / low) .
This could help in signal analysis.
By the way, that's a great job and so easy to use.
Best regards, YMasquel.

hneemann, 2021-07-09:
It is not possible to load a file but you can copy&paste the data to a test case.

YMasquel, 2021-07-09:
Thank you for this information but after some brainstorming I think that the use of a ROM memory and a counter I can solve my problem. I'll study your proposal to see what is the simplest and the finest solution.

closed 2021-07-09 by YMasquel

hneemann, 2021-07-10:
The use of a ROM/counter is of course also possible, but a test case is much more flexible and powerful, as it allows programming of the input signals to a certain degree. Whereas in a ROM only fixed values can be stored.