hneemann/Digital #789: Cannot Get External Component Working

AddioElectronics, 2021-07-11:
I'm just trying to get an external component to work with a very basic Verilog program.
GHDL and Iverilog is installed, but when I press the Check button they both give me errors.

I have 1 input named "clk," and 1 output named "q."

Here is my code.

```
module Test(clk, q);

input clk;
output q;

reg[3:0] counter;

reg q_r;

always @(posedge clk) begin
	if(counter < 15) begin
		counter <= counter + 1;
	end else begin
		q_r <= !q_r;
		counter <= 0;
	end
end

assign q = q_r;

endmodule
```

Here is the error I get when I press Check with "Generic" selected.

```
Input and output definitions could not be created automatically.
Please check the settings!
Check Result:
Executable file "module" not found!
```

And the check error I get with "GHDL" is selected.

```
Input and output definitions could not be created automatically. 
Please check the settings! 
Check Result:
Application exit status was not 0 but 1: 
Test.vhdl:1:1: missing entity, architecture, package or configuration module Test(clk, q); 
^ 
Test.vhdl:3:1: missing entity, architecture, package or configuration input clk; 
^ 
Test.vhdl:4:1: missing entity, architecture, package or configuration output q; 
^ 
Test.vhdl:6:1: missing entity, architecture, package or configuration reg[3:0] counter; 
^ 
Test.vhdl:6:8: bad extended digit reg[3:0] counter; ^ Test.vhdl:8:1: missing entity, architecture, package or configuration reg q_r; 
^ 
Test.vhdl:10:1: missing entity, architecture, package or configuration always @(posedge clk) begin 
^ 
Test.vhdl:13:9: missing entity, architecture, package or configuration end else begin 
^ 
Test.vhdl:15:17: missing entity, architecture, package or configuration counter <= 0; 
^ 
Test.vhdl:16:9: missing entity, architecture, package or configuration end 
^ 
Test.vhdl:22:1: missing entity, architecture, package or configuration endmodule 
^ 
C:\MinGW\bin\ghdl.exe: compilation error
```

And lastly here is the Check error I get for "IVerilog."

```
Check Result:
Application exit status was not 0 but -1,073,741,701:
```

In the options tab I have not added anything, but "--std=08 --ieee=synopsys" was under GHDL by default.

Any help to getting the external component working would be greatly appreciated.

Maybe if I  could suggest adding a small example to the documentation?
I'm not sure if I'm missing something, but there does not seem to be anything explaining how to actually use it.

Oh and thank you for creating such an amazing program, I just found out about it last night.

hneemann, 2021-07-11:
On my machine, the simulation starts without errors, but the code does not work.

Replacing

``` verilog
reg[3:0] counter;

reg q_r;
```

by

``` verilog
reg[3:0] counter=0;

reg q_r=0;
```
makes the code work as expected.

AddioElectronics, 2021-07-17:
Hmm interesting. I am still getting all those errors.

I will try compiling GHDL and IVerilog from source instead of using the precompiled binaries.

Or could it be that I'm using the 64-bit executables instead of the 32? 
I am on Windows.

hneemann, 2021-07-17:
Hmm! 
Which iverilog version are you using? Maybe something has changed there.

AddioElectronics, 2021-07-22:
mingw-w64-x86_64-iverilog-1_11.0.r8897.d8cb29f6-1-any.pkg.tar.zst
and I also tried
mingw-w64-i686-iverilog-1_11.0.r8897.d8cb29f6-1-any.pkg.tar.zst
from
https://ftp.osuosl.org/pub/msys2/mingw/mingw64/

For GHDL I downloaded it from https://github.com/ghdl/ghdl/releases

mingw-w64-x86_64-ghdl-llvm-ci-1-any.pkg.tar.zst
mingw-w64-i686-ghdl-mcode-ci-1-any.pkg.tar.zst

I put the executables in C:\MinGW\bin and C:\MinGW\mingw32\bin.

Just tried the 32-bit version, and I am getting a new error for the IVerilog Check.

```
Check Result:
error: Failed to open 'C:\MinGW\mingw32\lib\ivl\system.vpi' because:
: The specified module could not be found.
```

But system.vpi is certainly there. 
So I switched back to the 64-bit IVerilog, and tried renaming the 64-bit system.vpi to see if it would produce the same error, and while running the check, it is still giving me the first error I was getting(Below). So I'm guessing its exiting before it even tries to use system.vpi.

```
Check Result:
Application exit status was not 0 but -1,073,741,701:
```

hneemann, 2021-07-22:
Please try the following:

1. Create a circuit not containing a external component.
1. Add a test case to your circuit.
1. Create one or two test lines in the test case.
1. Make sure the tests are running in the simulation
1. Then export the circuit to verilog

Now a file will also be created to run the test.
You can run this test code from the command line.
Does the problem also appear on the command line when Digital itself is no longer involved?