hneemann/Digital #790: Any file/folder created by Digital are invisible for Windows

j-greffe, 2021-07-11:
Any file or folder I create with Digital is impossible to find with Windows Explorer as if they do not exist.
The files are accessible fine via Digital.

Here is version I use:
[image]

hneemann, 2021-07-18:
Digital creates the files without affecting the visibility. If files are surprisingly invisible, it is probably due to some security measures of the operating system.

hneemann, 2021-09-04:
@j-greffe 
Were you able to find out what happened?

j-greffe, 2021-09-06:
No luck with this.
I tried Digital on another Windows 10 and a Linux machine. It worked perfectly.
I suspect a file permission problem on this particular machine.
The only other clue that I have is that when I tried to save my files directly onto an USB key, I got the message that I did not have sufficient rights to write on it. I could write files on this USB key just fine with Windows Explorer.
Starting Digital as administrator did not resolve the issue either.