hneemann/Digital #794: Outputs spaced differently

IvyBit, 2021-07-17:
I'm propably just missing something but it seems that inputs in a sub circuit (shape: layout) are grouped and outputs get spaced evenly now. Is there an option I'm missing?

hneemann, 2021-07-18:
Inputs and outputs should behave the same.
Do you have an example circuit that shows this problem?

IvyBit, 2021-07-18:
I've checked again and the issue is not related to the type of port.
[pixel-counter.zip](https://github.com/hneemann/Digital/files/6836654/pixel-counter.zip)

[image]

IvyBit, 2021-07-18:
... I'm an idiot. 
Sorry, of course the outputs get spaced evenly unless there's not enogh space to do so.

IvyBit, 2021-07-21:
[image]
Something is still weird, this is how the project opened up today.

IvyBit, 2021-07-21:
Again.. the second I posted this I knew what was happening.
Windows server can have a nasty habit of caching files and forgetting them.

closed 2021-07-21 by IvyBit