hneemann/Digital #795: Errors during build process with mvn

tkoenig1, 2021-07-18:
Running "mvn install" from a clean build got me a few error messages:

```[INFO] Tests run: 5, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.133 s - in de.neemann.digital.cli.CommandLineTesterTest
[INFO] 
[INFO] Results:
[INFO] 
[ERROR] Errors: 
[ERROR]   VHDLSimulatorTest.testGHDLInSimulator:126 »  errors testing files: 5
[ERROR]   VHDLSimulatorTest.testInSimulator:46 »  errors testing files: 1
[ERROR]   VHDLSimulatorTest.testProcessorInSimulator:89->checkVHDLExport:143->runGHDL:159->startProcess:176 » IO
[INFO] 
[ERROR] Tests run: 800, Failures: 0, Errors: 3, Skipped: 0
[INFO] 
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  02:28 min
[INFO] Finished at: 2021-07-18T10:16:36+02:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:3.0.0-M1:test (default-test) on project digital: There are test failures.
[ERROR] 
[ERROR] Please refer to /home/ig25/Downloads/Digital/target/surefire-reports for the individual test results.
[ERROR] Please refer to dump files (if any exist) [date].dump, [date]-jvmRun[N].dump and [date].dumpstream.
[ERROR] -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
```
The errors themselves, scrolling back, were (with a bit of context)
```
../../src/synopsys/std_logic_arith.vhdl:255:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).

[ERROR] Tests run: 7, Failures: 0, Errors: 3, Skipped: 0, Time elapsed: 39.455 s <<< FAILURE! - in de.neemann.digital.hdl.vhdl2.VHDLSimulatorTest
[ERROR] testGHDLInSimulator(de.neemann.digital.hdl.vhdl2.VHDLSimulatorTest)  Time elapsed: 0.19 s  <<< ERROR!
java.lang.Exception: errors testing files: 5
        at de.neemann.digital.hdl.vhdl2.VHDLSimulatorTest.testGHDLInSimulator(VHDLSimulatorTest.java:126)

[ERROR] testInSimulator(de.neemann.digital.hdl.vhdl2.VHDLSimulatorTest)  Time elapsed: 21.695 s  <<< ERROR!
java.lang.Exception: errors testing files: 1
        at de.neemann.digital.hdl.vhdl2.VHDLSimulatorTest.testInSimulator(VHDLSimulatorTest.java:46)

[ERROR] testProcessorInSimulator(de.neemann.digital.hdl.vhdl2.VHDLSimulatorTest)  Time elapsed: 0.95 s  <<< ERROR!
java.io.IOException: 
Application exit status was not 0 but 1:
        ../../src/openieee/v93/numeric_std-body.vhdl:192:13:@0ms:(assertion warning): NUMERIC_STD.TO_01: non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:88:7:@0ms:(assertion warning): NUMERIC_STD.TO_INTEGER: non logical value detected, returning 0
../../src/synopsys/std_logic_arith.vhdl:255:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es)
```
I also found a few more errors, upon rerunning:
```
[main] INFO de.neemann.digital.hdl.vhdl2.VHDLCreator - export main
-- tested 51 examples
-- tested 5 examples
errors: 5
----> error in: /home/ig25/Downloads/Digital/src/test/resources/dig/external/ghdl/ghdl.dig
de.neemann.digital.core.NodeException: Could not create the external process!
Affected are: External.
        at de.neemann.digital.core.extern.External.init(External.java:116)
        at de.neemann.digital.draw.model.ModelCreator.createModel(ModelCreator.java:306)
        at de.neemann.digital.integration.ToBreakRunner.<init>(ToBreakRunner.java:68)
        at de.neemann.digital.integration.ToBreakRunner.<init>(ToBreakRunner.java:57)
        at de.neemann.digital.hdl.vhdl2.VHDLSimulatorTest.checkVHDLExport(VHDLSimulatorTest.java:132)
        at de.neemann.digital.hdl.vhdl2.VHDLSimulatorTest.lambda$testGHDLInSimulator$0(VHDLSimulatorTest.java:123)
        at de.neemann.digital.integration.FileScanner.scanIntern(FileScanner.java:81)
        at de.neemann.digital.integration.FileScanner.scan(FileScanner.java:39)
        at de.neemann.digital.hdl.vhdl2.VHDLSimulatorTest.testGHDLInSimulator(VHDLSimulatorTest.java:126)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.base/java.lang.reflect.Method.invoke(Method.java:566)
        at junit.framework.TestCase.runTest(TestCase.java:177)
        at junit.framework.TestCase.runBare(TestCase.java:142)
        at junit.framework.TestResult$1.protect(TestResult.java:122)
        at junit.framework.TestResult.runProtected(TestResult.java:142)
        at junit.framework.TestResult.run(TestResult.java:125)
        at junit.framework.TestCase.run(TestCase.java:130)
        at junit.framework.TestSuite.runTest(TestSuite.java:241)
        at junit.framework.TestSuite.run(TestSuite.java:236)
        at org.junit.internal.runners.JUnit38ClassRunner.run(JUnit38ClassRunner.java:90)
        at org.apache.maven.surefire.junit4.JUnit4Provider.execute(JUnit4Provider.java:365)
        at org.apache.maven.surefire.junit4.JUnit4Provider.executeWithRerun(JUnit4Provider.java:273)
        at org.apache.maven.surefire.junit4.JUnit4Provider.executeTestSet(JUnit4Provider.java:238)
        at org.apache.maven.surefire.junit4.JUnit4Provider.invoke(JUnit4Provider.java:159)
        at org.apache.maven.surefire.booter.ForkedBooter.invokeProviderInSameClassLoader(ForkedBooter.java:384)
        at org.apache.maven.surefire.booter.ForkedBooter.runSuitesInProcess(ForkedBooter.java:345)
        at org.apache.maven.surefire.booter.ForkedBooter.execute(ForkedBooter.java:126)
        at org.apache.maven.surefire.booter.ForkedBooter.main(ForkedBooter.java:418)
Caused by: java.io.IOException: Application exit status was not 0 but 1:
        /usr/bin/ghdl-mcode:warning: ieee library directory '/usr/lib/ghdl/mcode/vhdl/synopsys/v08/' not found
add.vhdl:3:10: unit "std_logic_unsigned" not found in library "ieee"
add.vhdl:3:10:  (use --ieee=synopsys for non-standard synopsys packages)
add.vhdl:14:26: entity 'add' was not analysed
add.vhdl:29:34: entity 'stdiointerface' was not analysed
```
with a few more errors like that.

tkoenig1, 2021-07-18:
This is on Ubuntu 20.04 LTS.  ghdl is:
```
$ ghdl --version
GHDL 0.37 (Ubuntu 0.37+dfsg-1ubuntu1) [Dunoon edition]
 Compiled with GNAT Version: 9.3.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2020 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

hneemann, 2021-07-18:
Hmm! I'm still using v0.37-dev.
Maybe there are some changes in the final version. I have to look at it.

hneemann, 2021-07-18:
I tested with v0.37 and all tests passed.
Maybe your installation is not complete. One of the error messages says that libraries are not found.

tkoenig1, 2021-07-18:
This is standard ubuntu ghdl, with no modification. There is a possibility of this being buggy. 

However, there is also an error message which might also be the cause of this build failure:

```
add.vhdl:3:10: unit "std_logic_unsigned" not found in library "ieee"
add.vhdl:3:10:  (use --ieee=synopsys for non-standard synopsys packages)
```
The ghdl documentation states

```
IEEE library pitfalls
=====================

When you use options :option:`--ieee=synopsys <--ieee>`, the ``ieee`` library contains non standard packages such as
``std_logic_arith``.
These packages are not standard because there are not described by an IEEE standard, even if they have been put in the
`IEEE` library.
Furthermore, they are not really de-facto standard, because there are slight differences between the packages of Mentor
and those of Synopsys.
Furthermore, since they are not well thought out, their use has pitfalls.
```
The missing file being complained about is also some synopsis stuff, so that might be related.

tkoenig1, 2021-07-21:
Just wondering... what do you get when you `ghdl --disp-config` ?  Maybe there is an option set
in your build environment that pulls in `std_logic_arith` somehow?

I get
```
$ ghdl --disp-config
command_name: /usr/bin/ghdl-mcode
command line prefix (--PREFIX): (not set)
environment prefix (GHDL_PREFIX): (not set)
exec prefix (from program name): /usr

library prefix: /usr/lib/ghdl/mcode
library directory: /usr/lib/ghdl/mcode/vhdl
default library paths:
 /usr/lib/ghdl/mcode/vhdl/ieee/v93/
 /usr/lib/ghdl/mcode/vhdl/
```

johnterickson, 2022-09-24:
I am hitting this in Ubuntu 20, but not on 22.  I think it may have been addressed as part of https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=929656