hneemann/Digital #799: Allow "Run to Break" to be interrupted

ProxyPlayerHD, 2021-07-22:
it happens to me quite a few times where i'm working on a circuit, haven't saved in a while, and then go to test it using the Break component because i want to look at some specific timings.
i place it, set the maximum cycles, and then press the "Run to Break" button.

and around 5 nanoseconds later i remember that the wire i connected the Break to will never activate.

the program is completely unresponsive while it's doing the simulation so i always have to either wait it out or kill the program using the Task Manager.

it would be very nice if i could just press the Stop button or ESC to immediately stop the "Run to Break" function.

hneemann, 2021-07-25:
Is releated to  #83

I really have to address this issue!

hneemann, 2021-07-31:
@ProxyPlayerHD 
What number of timout cycles have you set in the break component?

ProxyPlayerHD, 2021-07-31:
depends on the circuit.
but usually between 1M to 100M cycles.
i put them that high because otherwise i would have to do small adjustments, run the simulation, it didn't reach the point, do another small adjustment, repeat.

hneemann, 2021-07-31:
@ProxyPlayerHD 
Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/) (v0.27-88). While I'm not sure I like the implementation that much, it should work.

ProxyPlayerHD, 2021-08-01:
hmm i don't see any difference, the program still freezes once you click "run to break"

hneemann, 2021-08-01:
@ProxyPlayerHD 
That surprises me! Can you upload a minimal example where RunToBreak is not to be cancelled?

ProxyPlayerHD, 2021-08-02:
is uploading a circuit needed?
[image]
the counter is set to be 64 bits wide, and the Break component is set to 10M. it doesn't take long to reach the limit due to the simple circuit, but it'S still not interruptable while running.

hneemann, 2021-08-02:
@ProxyPlayerHD 
This is the exact circuit I use for testing and it works just fine!
Are you sure you are using a pre-release v0.27-88 or later?

ProxyPlayerHD, 2021-08-02:
the about page says "Build git-Revision: v0.27-88-g14f7168"

and just to clarify, so we're not accidentally talking about 2 different things, what exactly should you be able to do when you press "run to break"?
i assumed this would allow you to still click things in the menu bar, ie stopping the simulation and such, while the "run to break" is doing it's thing

hneemann, 2021-08-02:
> I assumed this would allow you to still click things in the menu bar, ie stopping the simulation and such, while the "run to break" is doing it's thing

exactly!

[image]

ProxyPlayerHD, 2021-08-02:
i was always using single gate mode, not regular simulation mode

hneemann, 2021-08-02:
@ProxyPlayerHD 
You use the single gate mode? 
That's interesting! For what purpose?

ProxyPlayerHD, 2021-08-02:
for example to see exactly when Outputs activate/deactivate or when Registers load in data.

or to single step through a CPU executing Instructions, so i can see each memory access. and such.

kinda a shame you can't look into sub circuits while the simulation is running.

hneemann, 2021-08-02:
> it's important to see how all the output signals do their thing (with the data graph) since most circuits i make i put onto programmable logic like CPLDs or FPGAs.

Keep in mind that the exact timing at gate level in the simulation has little or nothing to do with the later timing in an FPGA or a CPLD at this level of detail!
The single gate mode is only meant to show students which influence the timing on gate level can have. The exact timing on gate level in a FPGA or CPLD will be completely different!

ProxyPlayerHD, 2021-08-02:
it's a good enough estimate for the ~25MHz i'm running them at. and so far i never came across any noticable difference between what the simulation was doing and how the circuit worked on the actual hardware.

and in the worst case, i recently bought a Rigol DS1054Z Oscilloscope and did the "hack" to upgrade it to 100MHz. so i can always just look at the actual signals in real time.

hneemann, 2021-08-02:
It is just important to know that you can not rely on the simulation in this regard.
I think you are generally well advised not to rely on a specific timing at gate level! 
It is more robust to rely only on the timing at clock level.

ProxyPlayerHD, 2021-08-02:
either way the problem itself still remains.
single gate mode cannot be stopped while "run to break" is running...

hneemann, 2021-08-02:
> either way the problem itself still remains.

That's right!  :smile: 
Please try [v0.27-94](https://infdigital.dhbw-mosbach.de/)

ProxyPlayerHD, 2021-08-03:
alright that seems to work, it's very laggy but that's still much much better than just straight up freezing the whole program.

hneemann, 2021-08-03:
> it's very laggy

Please try [v0.27-95](https://infdigital.dhbw-mosbach.de/). Still not perfect but better.

commit d889c247 (2021-08-03): attempt to fix #799
closed 2021-08-03 by hneemann