hneemann/Digital #801: Wishlist item: AOI

tkoenig1, 2021-07-27:
It would be interesting to have AOI component. It can, of course, be generated from existing components, but it something that can bring advantages in CMOS, so it would be nice to have, especially when counting components.

hneemann, 2021-07-28:
Wouldn't it only make sense if you count transistors?

tkoenig1, 2021-07-28:
Counting components as a prerequisite for counting transistors, certainly.

hneemann, 2021-07-29:
Hmm! 
But it is possible to build such a circuit from CMOS transistors. 
Then the transistors are counted correctly.

[image]

[21AOI.dig.zip](https://github.com/hneemann/Digital/files/6898206/21AOI.dig.zip)

closed 2021-08-07 by hneemann