hneemann/Digital #804: 74670 Component not working

hwreverse, 2021-08-06:
[image]

Hi there,

Long story short: Any circuit with the 74670 component does not run and gives the following error:

**No output connected to a wire([1;Splitter]). The state of the wire is undefined.
Occurred in file 74.670.dig**

Best regards,
Alexander

Sketch from the image attached to Issue report
[670d1.zip](https://github.com/hneemann/Digital/files/6946879/670d1.zip)

hneemann, 2021-08-06:
In Digital it is not allowed not to connect inputs.
See also  #36

hwreverse, 2021-08-06:
Thank you very much, it solved the problem.

closed 2021-08-06 by hwreverse

hwreverse, 2021-08-06:
As Digital did highlight the "VCC" net during the error, I was lead to believe (in error) that the error was related to this wire in particular.

hneemann, 2021-08-06:
Highlighting a wire in light green indicates a logic 1, which is of course always the case with VCC. In case of an error it is highlighted in red.
But in your case there were no connected wires that could be highlighted.

hwreverse, 2021-08-06:
That clears it up. Thank you very much for your fast response. And thank you for your great piece of software!