hneemann/Digital #819: Missing active low inputs/inverted outputs labels on 74xx chips

PaulKappmeyer, 2021-09-02:
Some 74xx chips indicate that an input is active low (inputs are labeled with ~).
I think that some chips are missing this indication. I just found those two:

- 74173 (Quad D-flip-flop): active low inputs: OE1 (pin 1), OE2 (pin2), DE1 (pin 9), DE2 (pin 10)
- 74189 (64-Bit RAM): inverted outputs: Q0 (pin 5), Q1 (pin 7), Q2 (pin 9), Q3 (pin 10)

I’m not quite sure if the labels also apply to inverted data outputs, but I think that it would be a good idea.
Also, what about a chip like 7404 (hex inverter) where technically every output is an inverted output. Are the labels unnecessary in this case or could it still be helpful?

hneemann, 2021-09-02:
The labels are usually taken from the datasheets and I would like to stick with that.
And although different manufacturers sometimes use different labels for the same chips, I would not like to create another variant.

PaulKappmeyer, 2021-09-02:
Okay, but I didn’t really mean another variant.

[image]

These are two chips that both have got an active low output-enable pin. Looking at the bottom one I can see that the input is active low because of the line above the pin-name (OE). This line seems to be missing at the top chip, doesn’t it?
But it is really easy to add this line by just opening the circuit, activating the modification and adding this ~ symbol before the pin-name.
[image]

I just did that for myself and wanted to ask if this shouldn’t be changed in the files of this repo? 

Looking at this datasheet for example (link: https://www.ti.com/lit/ds/symlink/sn74ls173a.pdf?ts=1630593202378&ref_url=https%253A%252F%252Fwww.google.com%252F):
They have named pins 1, 2 differently, but at least pins 9, 10 have the line above the pin name.
[image]

Sorry again if I misunderstood your answer, but this way it seemed a bit inconsequent.

hneemann, 2021-09-03:
Hmm! I have not created all the IC's myself. Maybe someone did not know about the ~ sign?

xotmatrix, 2021-09-03:
There is also the fact that datasheets are very inconsistent, even within one vendor's data book. That 74173 looks like it was taken from Motorola's 74HC173 datasheet where the labels appear just as they do in Digital, without overbars, and totally different than other vendor datasheets. Motorola's 74HC73, 74HC74A, 74HC173 flipflop datasheets all use different label names, with and without overbars or subscripts, and with different capitalization. Motorola's 74LS173A datasheet is also very different than its 74HC173 datasheet.

hneemann, 2021-09-03:
@xotmatrix 
Thank you for the effort to look at it in such detail. Under these circumstances, it is probably most reasonable to leave it as it is.
I also think that it does not make much difference, because who uses 74xxx chips will know the chips well or consult the data sheets anyway.

closed 2021-09-09 by PaulKappmeyer