hneemann/Digital #820: Chinese translation update

hneemann, 2021-09-04:
@itviewer 
This is a request for updating the language files for a planned new release.
I would appreciate your continued support.

[lang_diff_zh.xml.zip](https://github.com/hneemann/Digital/files/7109341/lang_diff_zh.xml.zip)

itviewer, 2021-09-06:
[lang_diff_zh.zip](https://github.com/hneemann/Digital/files/7113039/lang_diff_zh.zip)

hneemann, 2021-09-06:
@itviewer 
Thank you again for your help!

There were some syntax errors when importing the file. Can you take a look at it again?

```
zh: Param count does not match: err_errorWritingFile_N="写文件时发生错误。", expected: 1, found 0
zh: Param count does not match: err_file_N0_ExistsTwiceBelow_N1="文件 {0} 存在多次", expected: 2, found 1
zh: Param 1 is missing in: err_wrongNumOfArgsIn_N0_InLine_N1_found_N2_expected_N3="函数 {0} 的参数个数不正确（期望 {3} 个，发现 {2}个）"
zh: Missing closing '}': msg_fixesCreated_N="Fixtures: {0"

```

itviewer, 2021-09-06:
[lang_diff_zh.zip](https://github.com/hneemann/Digital/files/7113896/lang_diff_zh.zip)

hneemann, 2021-09-06:
This time it worked perfectly.
Thanks again for your help!

commit c4477534 (2021-09-06): Chinese translation update; closes #820
closed 2021-09-06 by hneemann