hneemann/Digital #823: Clock speed

PaulKappmeyer, 2021-09-09:
Is there a way to adjust the speed of the clock while the simulation is running?

hneemann, 2021-09-10:
So far, this is not possible.

PaulKappmeyer, 2021-09-13:
Do you think this could be added?
I don’t think that there are that many use cases, but I think it could be a nice feature still.

hneemann, 2021-09-16:
This is not quite so simple, since different implementations of the clock are used for different clock speeds. For example, at high clock speeds, the simulation runs in its own thread, fully utilizing one core, and the GUI thread's access is synchronized accordingly. It is not so easy to switch in between in the running simulation.
I don't think I want to add variable clock rates.

YulienPohl, 2021-09-18:
This problem can be solved by a variable clock divider containing a counter and a buffered output to avoid hazards.
[Variable Clock Divider.zip](https://github.com/hneemann/Digital/files/7190265/Variable.Clock.Divider.zip)

hneemann, 2021-09-18:
@YulienPohl 
What a simple solution to the problem! :smile:
I should have thought of it myself! Thanks for pointing this out! :+1:

closed 2021-09-18 by hneemann