hneemann/Digital #827: Support for n-ary input gates

nicola-gigante, 2021-09-14:
Hi and first of all thank you for the great work!

It would be useful if Digital could support logic gates with more than two inputs. Or am I missing how to have them already?

hneemann, 2021-09-14:
You can either point to the component and press '+', or increase the number of inputs in the component's attributes dialog.

closed 2021-09-14 by hneemann

nicola-gigante, 2021-09-14:
Sorry for bothering you, I'm new to Digital, but if I press + after clicking on the component nothing happens. And where's the component's attributes dialog?

xotmatrix, 2021-09-14:
If you are on Windows and + is a normally a shifted key (on my keyboard it is Shift-=) then try the key without pressing shift.
Right-click a component to bring up the attributes.

nicola-gigante, 2021-09-15:
The right click wasn't working for me so I never knew about the attribute dialog! I had to disable the "Use macOS mouse clicks" preference to make it work.

flandreas, 2021-09-15:
I'm on macOS as well, and for me CTRL + "Left Click" works with setting "Use macOS mouse clicks" enabled.

hneemann, 2021-09-15:
As far as I know, on the Mac traditionally the context menu is opened with Ctrl-Click, because traditionally a mouse with only one button was used. But nowadays everyone has a mouse with two buttons, so you can also use right-click, as it is common in Windows.
But for this the option "Use MacOS mouse clicks" must be disabled.

nicola-gigante, 2021-09-15:
Ok, got it. As you said, nowadays everybody uses a two-button mouse. Apple itself has selled two-button mice for nearly two decades now. So that option being the default sounds a bit anachronistic. May I suggest to disable it by default? Thanks anyway.

hneemann, 2021-09-15:
Maybe! 
However, this option only exists because users have wondered why the context menu cannot be opened with Ctrl-Left. :smile:

nicola-gigante, 2021-09-15:
Fair enough! But why are the two modes mutual exclusive? I mean, why can't both right click and Ctrl-Left clicks be supported at the same time?

hneemann, 2021-09-15:
Because Ctrl is traditionally the primary click modifier in Windows. See also  #136 or [here](https://github.com/hneemann/Digital/blob/master/src/main/java/de/neemann/gui/Mouse.java).

nicola-gigante, 2021-09-15:
Ok, got it!