hneemann/Digital #832: little endian versus big endian

d-teif, 2021-09-15:
Suppose the following hex file for mips program 
:04000000241D3FFC80
:040004000C000029C3

With EEPROM ou ROM 8 bits the content is OK that mean 0x24 0x1D 0x3F 0xFC

With EEPROM ou ROM 32 bits the content is 0xFC3F1D24. In this case I have to use a splitter to-reorder data in ROM to obtain good value 0x241D3FFC

I will be great to have little or big endian selection ?

Thanks

hneemann, 2021-10-02:
see also #522

hneemann, 2021-10-02:
@d-teif 
Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/) and let me know how it works.

d-teif, 2021-10-02:
Thanks, It's work perfectly. With Big endian button the program bytes are load in right order in 32 bits data ROM and EEPROM.

I noticed another problem. When you try to create 32 bits address memory, the value 32 seem valid, but real size of address is 24 bits.
Is 24 bits a maximum address size for memory components ? If I a remember well same thing occurs in logisim!

Thanks again

hneemann, 2021-10-02:
@d-teif 
In the simulator all data is stored in a long, i.e. 8 bytes. If 32 address bits were used, this would occupy 32 GBytes in the main memory. Therefore the number of address bits is limited to 24, which corresponds to 128 MByte. If only 32 bit words are used that would still be 64MByte of usable memory. 
What is the use case for which more than 64 MByte memory is needed in the simulation?

closed 2021-10-02 by hneemann

d-teif, 2021-10-02:
Hi used 32 addess bits in mips multicycle processor in my computer architecture material. Program is at 0x40000, data at 0x1001000 and pile donwto 0x7FFFEFFC in standard mips architecture. I use 3 memory , one rom for code, one eeprom for static data and one ram for stack! And It's ok, just wander why this 24 bit limitatation! 

Thanks again 

Didier

commit cfb7a191 (2021-10-02): Updates the release notes; closes #832