hneemann/Digital #833: exporting to verilog with multiple outputs connected

sulsadood, 2021-09-15:
Hi, I have been making a simple computer in Digital and using the generated verilog files for the qflow layout process. I have run into a problem where I tried exporting to verilog with multiple outputs connected and it would say that it isn't supported. I was able to fix this problem by replacing two tristate buffers/drivers with pin controls. However, even though it will export to verilog it fails at the placement step of the qflow process. I tried running individual components through the qflow process and discovered some problems which I was able to fix but my original issue still remains. Is there a different way I am suppose to be doing this?
[EIGHTbitCompA.zip](https://github.com/hneemann/Digital/files/7171310/EIGHTbitCompA.zip)

hneemann, 2021-09-15:
It is not possible to design your own d-flipflop with simple gates and export it to Verilog or VHDL. Digital is not able to recognize these self-created d-flipflops and automatically replace them with the built-in ones. And it is difficult for any synthesis tool to synthesize unbuffered loops.
Remember that an fpga is essentially made up of ready-made d-flipflops connected via look up tables.
If you use the built-in d-flipflops for your processor and do not build memory elements from simple gates with loopbacks, it will work.
Note that the d-flipflops are best driven by a common clock, although it might work if you don't follow this.

See also #184

sulsadood, 2021-09-15:
I went through and replaced all the basic components I made myself with the premade ones in Digital. I also looked through that other post, but that seems to be more specifically for FPGAs. I am using qflow just to make a layout. The loopback is what I am using as a bus. Is there a different way I can implement a bus like this?

sulsadood, 2021-09-15:
I also got rid of all the drivers and instead use a mux system to decide what to output to the bus, but I am still having the same issue. The parts individually have no issues, but when I try to connect them together and run that through qflow it cant do it. Do you have any more suggestions?

sulsadood, 2021-09-16:
I found the problem; it was because I was using the asynchronous ram module. I switched to separate port ram module and it worked like a charm.
[EIGHTbitCompA.zip](https://github.com/hneemann/Digital/files/7174843/EIGHTbitCompA.zip)

closed 2021-09-16 by sulsadood